NixOS/nixos-hardware #8: General Thinkpad config

Profpatsch, 2016-02-16:
First off, I really like the idea behind this repo.

We should abstract a bit more to find good optimizations that work for every thinkpad (or at least a category of them).

e.g. I factored out some in [vuizvui](https://github.com/openlab-aux/vuizvui):

```nix
{ lib, config, pkgs, ... }:

with lib;

let
  cfg = config.vuizvui.hardware.thinkpad;

in
{
  options.vuizvui.hardware.thinkpad = {
    enable = mkEnableOption "thinkpad support";
  };

  config = mkIf cfg.enable {
    # read acpi stats (e.g. battery)
    environment.systemPackages = [ pkgs.acpi ];

    # for wifi
    hardware.enableAllFirmware = true;

    hardware.trackpoint = {
      enable = true;
      emulateWheel = true;
      speed = 250;
      sensitivity = 140;
    };

    # TLP Linux Advanced Power Management
    services.tlp.enable = true;
  };
}
```

This piece of code also uses the power options give (e.g. I then use `vuizvui.hardware.thinkpad.enable` in my config). vuizvui is a bit special though, I don’t know if it’s viable to do with a drop-in module. @aszlig could tell us more.

ehmry, 2016-02-16:
Yes, I think we should have generic thinkpad modules that can live in the lenovo folder, and then locally import those into the model specific files. Someone else mentioned that they might be putting one together.

I would say drop the ``mkEnableOption`` and enable everything so long as the module is imported. I did not know about wheel emulation on the trackpoint until now, but I would not change the sensitivity by default.

ehmry, 2016-02-16:
Well actually, you could still use ``mkEnableOption`` for the vuizvui flavor, and seperate those options from the vanilla options.

Profpatsch, 2016-02-16:
> I did not know about wheel emulation on the trackpoint until now, but I would not change the sensitivity by default.

The default is simply the xorg default, which is probably worse for any Thinkpad.
Furthermore, when a user redefines it in his config, it will overwrite the values set in the imported config(s).

aszlig, 2016-02-18:
The reason I did that with `mkEnableOption` by convention is that having this available without the need to add another import is that we get a manual entry for it without to explicitly document it elsewhere. Of course all of this can still be factored out for modules that are "enable the whole module or nothing" instead of using a bunch of repetitive `mkEnableOption` declarations.

DamienCassou, 2016-08-07:
You might want to switch to `libinput` instead which takes care of weird configurations such as T450. Look at what I did in https://github.com/NixOS/nixos-hardware/blob/master/lenovo/t460s.nix#L45.

Profpatsch, 2016-08-08:
Yes, `libinput` seems like the way to go. There is a module, but it’s kind of broken and support for different devices is missing at the moment.
The manual method works.

Profpatsch, 2016-08-08:
Hm, `libinput` does provide an api for kinetic scrolling, but X doesn’t implement it.
So we should support both.

Ma27, 2017-09-03:
@Profpatsch any progress here? I originally wanted to contribute the tweaks I built for my ThinkPad x270 here, but it would mostly duplicate the other ThinkPad configurations.

If nobody works on that ATM, I'd love to give it a try :-)

Profpatsch, 2017-09-03:
@Ma27 Go for it!

Ma27, 2017-09-03:
see #25 for a first WIP :-)

Profpatsch, 2017-10-20:
#25 is merged, so the basics are working now.

closed 2017-10-20 by Profpatsch