NixOS/nixos-hardware #11: This repository should live underneath https://github.com/NixOS/

peti, 2016-04-20:
It's very hard for new users to discovers this repository. Also, it belongs into the organization from a structural point of view.

ehmry, 2016-04-20:
@peti I'm all for it but the problem is I don't have the rights to move it into nixos on my own.

ehmry, 2016-04-20:
If someone creates a new repo at NixOS I can remove this one.

peti, 2016-04-20:
Just contact Eelco or Rob in IRC or by email and ask  them to create a repo for you. That shouldn't be a problem?

ehmry, 2016-04-20:
@edolstra @rbvermaa how about it?

Ericson2314, 2016-04-20:
@ehmy Last I checked I thought anybody could transfer a repo, just need somebody in the org to accept transfer?

If that's not the case better to give an org member admin rights to this repo, so they can transfer it. That way fork graph and issue will be preserved.

ehmry, 2016-04-20:
Thats how it should work, but alas, not how it does work.

Ericson2314, 2016-04-20:
Then my plan B?

ehmry, 2016-04-20:
I can't give admin rights to a repo, but I could probably transfer the repo to a nixos admin who could then pass it on the the nixos org.

Ericson2314, 2016-04-20:
That will do. Thanks!

edolstra, 2016-04-21:
Hm, we should probably define the scope/purpose of this repository a bit. It appears to be two things:

* A repository of hardware-related modules (with some user configurations thrown in, e.g. https://github.com/ehmry/nixos-hardware/blob/master/inversepath/usbarmory/configuration.nix). When should hardware modules go into `<nixpkgs/nixos/modules>` and when into this repository? Putting hardware modules in a separate repository creates a version management problem, e.g. when a module refers to options/packages introduced in a particular NixOS revision. Also, it's hard for `nixos-generate-config` to refer to modules not in the Nixpkgs repo, while ideally hardware quirks would be detected automatically by `nixos-generate-config`.

* A wiki. But do we want to have a separate wiki *only* for model-specific hardware issues?

peti, 2016-04-21:
My understanding was that this repository is a loose collection of code snippets that people can cut&paste and/or import into their `configuration.nix`. Arguably, this is Wiki'ish in nature, only it's tracked in a proper Git repository.

ehmry, 2016-04-25:
Having detectable hardware modules in ``nixpkgs/nixos/modules`` would be preferable, but I think there is there is a grey area for quality assurance. That seems to make this a wiki of user snippets, but what is the plan for the https://nixos.org/wiki ? @garbas any thought?

Ericson2314, 2016-04-25:
Yeah, if this really takes off, we can
1. Merge it in nixpkgs (remember `git subtree` can help preserve history)
2. Integrate the Nixos modules with existing hardware detection. (As far as QA is concerned, we'd could decide whether to use them by default---boils down to a kind of parametricity property.)
3. Move the wikis with the rest of our documentation, also the destination of the old wiki's info last I checked.

But given that this is still an experiment, and sticking even more things in nixpkgs would require some sort of community consensus, I think moving this repo is the proper first step for now.

edolstra, 2016-04-25:
Yeah I agree. I've created https://github.com/NixOS/nixos-hardware, which should be writable to all Nixpkgs committers.

Ericson2314, 2016-04-25:
@edolstra did you see my earlier comments about moving the repository so github fork information and other metadata is preserved?

edolstra, 2016-04-26:
Sorry, didn't see that. So what should I do to move the repo?

ehmry, 2016-04-26:
Meh, I already pushed to the new one.

ehmry, 2016-04-26:
@edolstra I transfered the repo to you, so you should be able to hand it off to the Nixos org.

edolstra, 2016-04-26:
Okay, done. Thanks!

closed 2016-04-26 by edolstra

Ericson2314, 2016-04-26:
Excellent; thank you all!

garbas, 2016-04-28:
thank you! this reminds me i have to put in my lenovo x250 configs there :)