NixOS/nixos-hardware #15: The description of this repo links to this repo

siddharthist, 2016-07-13:
Everyone who sees this page will know where it is ;)

copumpkin, 2016-08-31:
Yeah, seems a little odd 😄

primeos, 2017-01-25:
This actually confuses me every time...

It feels like being at the wrong place ("Moved to") and like I'm supposed to click that link. Would be nice if we could get rid of it :smile:

And it doesn't blend in with the other descriptions: https://github.com/NixOS

We could change it to "A collection of NixOS modules covering hardware quirks." from the README.md for example :wink:

cc @ehmry

ehmry, 2017-01-25:
I don't have the rights, methinks it takes a NixOS admin, @garbas ?

garbas, 2017-01-25:
fixed the description thx @primeos for making me not think what to write :)

@ehmry you should have the admin right for this repo now.

closed 2017-01-25 by garbas