NixOS/nixos-hardware #16: What’s the hardwareNotes option for?

Profpatsch, 2016-07-25:
https://github.com/NixOS/nixos-hardware/blob/master/lib/hardware-notes.nix#L21

Needs a description, it’s not very obvious what this is for (apart maybe from searching for examples).

yegortimoshenko, 2017-12-22:
I think `hardwareNotes` should be phased out in favor of per-profile README file like this one: https://github.com/NixOS/nixos-hardware/blob/e5cc44520fecf4b257f33cd6acf615d528c2a110/apple/macbook-pro-10-1.wiki

It largely has the same goal. `hardwareNotes` is not very useful in that it can only be comfortably read after system is installed, and doesn't offer any formatting.

zimbatm, 2017-12-22:
Good idea. I don't know what the `.wiki` format is so maybe something more common like `.md`. The content can still be later pulled into the nix code with `builtins.readFile`

yegortimoshenko, 2017-12-22:
`.wiki` is just MediaWiki format, like Wikipedia articles. I think it might have been copied from old NixOS wiki. I don't think there is much point in enforcing single format, at least not until this repo sees more usage. Markdown sometimes is problematic in that tables are messy, and it's very poorly standardized.

zimbatm, 2017-12-22:
Tables in Github's Flavored Markdown are reasonable enough:

| h1 | h2 |
| - | - |
| value1 | value2 |

In any case, don't let me hold you back if you want to convert everything to .wiki or .org. There is always `pandoc` to do the heavy lifting in case we want to change later on.

yegortimoshenko, 2017-12-22:
I'd rather not convert anything for now :-) Just leaving it as it is. Probably we'll have to convert it at some point.  I've used Org mode in #29 because it handles tables really well and has and hides links when they are used, feels almost like editing rich text.

commit e1806b4c (2017-12-24): lib: drop, fixes #16 (#37)

Hardware notes are seemingly less useful than a README, and
kernelAtLeast option, while potentially useful:

a) is not used anywhere in the tree at the moment,
b) doesn't properly work when there are multiple elements in the list,
c) assumes that there are no degradations ever.
closed 2017-12-24 by zimbatm