NixOS/nixos-hardware #31: lenovo: Review (improve profile quality)

yegortimoshenko, 2017-12-22:
https://github.com/NixOS/nixos-hardware/blob/440088677b4e6f9527466cbc5b02abf2bf3f863d/lenovo/thinkpad/x220i-tablet.nix#L9-L13

`tp_smapi` is not specific to this particular model, but works with ThinkPads in general: http://www.thinkwiki.org/wiki/Tp_smapi#Model-specific_status

https://github.com/NixOS/nixos-hardware/blob/440088677b4e6f9527466cbc5b02abf2bf3f863d/lenovo/thinkpad/x220i-tablet.nix#L14-L15

`hdapsd` should be used for all laptops that have a hard drive, and any ThinkPad model might use SSD instead of a hard drive since it's not soldered in.

https://github.com/NixOS/nixos-hardware/blob/440088677b4e6f9527466cbc5b02abf2bf3f863d/lenovo/thinkpad/x220i-tablet.nix#L17-L18

`fprintd` is commented out and thus not available if profile is read-only (#29). Fingerprint reader might or might not be available depending on hardware configuration.

https://github.com/NixOS/nixos-hardware/blob/440088677b4e6f9527466cbc5b02abf2bf3f863d/lenovo/thinkpad/t460s.nix#L21-L22

Whether to use EFI boot or legacy boot and which bootloader to use should not be coupled with hardware configuration, in my opinion, for all devices except where it's explicitly coupled, like Raspberry Pi.

https://github.com/NixOS/nixos-hardware/blob/440088677b4e6f9527466cbc5b02abf2bf3f863d/lenovo/thinkpad/t410.nix#L35-89

Either POWERTOP or tlp should handle this.

https://github.com/NixOS/nixos-hardware/blob/440088677b4e6f9527466cbc5b02abf2bf3f863d/lenovo/thinkpad/t410.nix#L33

This should be on for all hardware that uses Intel processors, unless user explicitly opts out.

https://github.com/NixOS/nixos-hardware/blob/440088677b4e6f9527466cbc5b02abf2bf3f863d/lenovo/thinkpad/t410.nix#L7-L26

Some options seem to be too aggressive, which goes against README saying that stability should be favored above all else.

https://github.com/NixOS/nixos-hardware/blob/440088677b4e6f9527466cbc5b02abf2bf3f863d/lenovo/thinkpad/general-intel.nix#L1-L4

This should be on for every machine that has a modern Intel processor with integrated GPU.

---

Larger problem is that ThinkPads don't usually warrant a per-model hardware profile. I still think it is useful to have a per-model hardware profile in case if things have to be fixed later on, but most configuration in these profiles could be just moved to `common.nix`.

zimbatm, 2017-12-22:
You make a lot of good observations. Some further thoughts:

It's not clear against which nixpkgs checkout there profiles are tested. Some defaults might be changing between releases (for example if you make a case for enabling the microcode updates by default). In general I think that the rule should be that the profile should work both on the latest stable release. And optionally on master.

I don't think that we should be generalizing anything in this repo. Either it's a setting that applies across the board and nixpkgs should be changed, or it gets duplicated into individual profiles. Otherwise we will quickly end-up in a situation where users want to change a value but have no way to test if it works on the other platforms. The amount of coordination would be insane. As developers we are taught to factor out duplication but I think that rule should be avoided here. copy-and-paste is the answer.

A corollary of the last paragraph is that we should introduce a notation to say which github handle maps to which configuration. When changing a specific config, ping the other people marked in the config for approval.

yegortimoshenko, 2017-12-22:
> for example if you make a case for enabling the microcode updates by default

It's not enabled by default (as a NixOS option) because not everyone has an Intel CPU. That doesn't explain why it isn't generated by `nixos-generate-config`, though, but probably just not implemented (I'll open an issue).

> Otherwise we will quickly end-up in a situation where users want to change a value but have no way to test if it works on the other platforms. 

Then they can change it locally, i.e. per-profile or per-system. Currently `lenovo/` reads more like specific user configs cobbled together and not something that can just be imported to have a working machine.

All I want is precisely what has been stated in the README: workarounds for hardware quirks.

> It's not clear against which nixpkgs checkout there profiles are tested.

I think this repo just should have a release branches like Nixpkgs that will only get fixes. I think `18.03` should be tagged and `release-18.03` should be branched out when it's ready.

zimbatm, 2017-12-22:
Agreed on the general sentiment. To introduce these changes I recommend to make separate PRs for each of them and @mention the committers of these profiles.

yegortimoshenko, 2017-12-24:
> Agreed on the general sentiment. To introduce these changes I recommend to make separate PRs for each of them and @mention the committers of these profiles.

Sure, will do, on any breaking changes.

yegortimoshenko, 2017-12-25:
This has been mostly resolved, except for boot loader options that I've left for reverse compatibility. I'll open a PR that removes boot options.

closed 2017-12-25 by yegortimoshenko