NixOS/nixos-hardware #33: undefined variable

Nichlas, 2017-12-23:
When I try to add Thinkpad T410 to my config i get the error: `error: undefined variable ‘mkDefault’ at /nix/var/nix/profiles/per-user/root/channels/nixos-hardware/lenovo/thinkpad/intel.nix:2:24`

I see that the variable is defined in the file `common.nix` but it doesn't look like this file is included in the profile or in  `intel.nix` that uses the variable.

zimbatm, 2017-12-24:
looks like you broke something @yegortimoshenko

yegortimoshenko, 2017-12-24:
@Nichlas Fixing right away. Sorry.

closed 2017-12-24 by yegortimoshenko

zimbatm, 2017-12-24:
@yegortimoshenko that's what I meant by making changes on hardware you don't own. It happens to the best of us.

yegortimoshenko, 2017-12-24:
@zimbatm I'll test these profiles next time. I have very similar hardware and honestly none of ThinkPad profiles currently in the tree are specific to any ThinkPad model.

yegortimoshenko, 2017-12-24:
Also we really need evaluation tests (#7).

yegortimoshenko, 2017-12-24:
I've also subscribed to this repo and will get a notification on issues. Thanks for pinging me!

yegortimoshenko, 2017-12-24:
@Nichlas: current `t410` profile uses potentially unstable GPU settings. Could you test it out and tell how stable it is, how's battery life, etc. Thanks!

Nichlas, 2017-12-24:
Yes, I'm plannig to give it at go.

currently got another error though `error: undefined variable ‘config’ at /nix/var/nix/profiles/per-user/root/channels/nixos-hardware/lenovo/thinkpad/default.nix:6:32
(use ‘--show-trace’ to show detailed location information)`

yegortimoshenko, 2017-12-24:
Fixed, please update the channel: `sudo nix-channel --update nixos-hardware`.

commit b8ab6ab3 (2017-12-24): thinkpad/intel: drop mkDefault, fixes #33