NixOS/nixos-hardware #41: Link with the ArchLinux wiki

zimbatm, 2017-12-25:
The [ArchLinux wiki](https://wiki.archlinux.org/) is a wonderful source of information. It has a great deal of information on hardware support and is actively kept up2date. Let's piggy-back on it.

I can think of two things that could be useful:
* profile naming: there exists a lot of hardware out there. When in doubt with the profile name, that the name of the ArchLinux wiki page.
* profile info: link back to the ArchLinux wiki page. This will be useful to motivate various configuration decisions.

I think that we should piggy-back on it. For example if there is an ArchLinux wiki page for a specific hardware

zimbatm, 2017-12-25:
Also I quite like the hardware report that they are giving. Eg: https://wiki.archlinux.org/index.php/Lenovo_ThinkPad_Edge_E455 . It gives a good idea of what has been tested to work.

yegortimoshenko, 2017-12-25:
I agree, we should link to it in per-model README file whenever it makes sense to do so, but there are other great resources that we should also link to, [ThinkWiki](https://www.thinkwiki.org/wiki/ThinkWiki) for ThinkPads, [EveryMac](https://everymac.com/) for Macs...

Also, the point of this thing is so that people don't have to configure anything themselves, so we shouldn't rely on the user to check the ArchWiki or any other resource. This should be mostly for maintenance.

Mic92, 2021-06-13:
I think we already do this where appropriate. So let's close it.

closed 2021-06-13 by Mic92