NixOS/nixos-hardware #49: Identifying devices through DMI ID?

samueldr, 2018-01-16:
The files under `/sys/devices/virtual/dmi/id/` should allow most laptops, and probably brand-name pre-built computers too.

This could probably be used to detect available hardware configuration in some way, and the values could be used to help categorize the files in a way that is automatically searchable. Some of that information (modalias IIRC) is used in the udev hwdb to detect hardware and apply quirks to them.

I have attached a dump of the data in different classes of machines I own. This will help kickstart discussion and discovery of the helpful attributes.

* * *

##### Acer C720p (with MrChromebox's coreboot-based UEFI)

```
~ $ (cd /sys/devices/virtual/dmi/id/ ; for f in *; do echo -n "$f="; cat "$f"; done)
bios_date=07/14/2017
bios_vendor=coreboot
bios_version=MrChromebox
board_name=Peppy
board_serial=cat: board_serial: Permission denied
board_vendor=GOOGLE
board_version=1.0
chassis_asset_tag=
chassis_serial=cat: chassis_serial: Permission denied
chassis_type=9
chassis_vendor=GOOGLE
chassis_version=
modalias=dmi:bvncoreboot:bvrMrChromebox:bd07/14/2017:svnGOOGLE:pnPeppy:pvr1.0:rvnGOOGLE:rnPeppy:rvr1.0:cvnGOOGLE:ct9:cvr:
power=cat: power: Is a directory
product_family=Google_Slippy
product_name=Peppy
product_serial=cat: product_serial: Permission denied
product_version=1.0
subsystem=cat: subsystem: Is a directory
sys_vendor=GOOGLE
uevent=MODALIAS=dmi:bvncoreboot:bvrMrChromebox:bd07/14/2017:svnGOOGLE:pnPeppy:pvr1.0:rvnGOOGLE:rnPeppy:rvr1.0:cvnGOOGLE:ct9:cvr:
```

##### ASUS TP300LA

```
~ $ (cd /sys/devices/virtual/dmi/id/ ; for f in *; do echo -n "$f="; cat "$f"; done)
bios_date=11/09/2015
bios_vendor=American Megatrends Inc.
bios_version=TP300LA.205
board_asset_tag=ATN12345678901234567
board_name=TP300LA
board_serial=cat: board_serial: Permission denied
board_vendor=ASUSTeK COMPUTER INC.
board_version=1.0       
chassis_asset_tag=No Asset Tag    
chassis_serial=cat: chassis_serial: Permission denied
chassis_type=10
chassis_vendor=ASUSTeK COMPUTER INC.
chassis_version=1.0       
modalias=dmi:bvnAmericanMegatrendsInc.:bvrTP300LA.205:bd11/09/2015:svnASUSTeKCOMPUTERINC.:pnTP300LA:pvr1.0:rvnASUSTeKCOMPUTERINC.:rnTP300LA:rvr1.0:cvnASUSTeKCOMPUTERINC.:ct10:cvr1.0:
power=cat: power: Is a directory
product_family=TP
product_name=TP300LA
product_serial=cat: product_serial: Permission denied
product_uuid=cat: product_uuid: Permission denied
product_version=1.0       
subsystem=cat: subsystem: Is a directory
sys_vendor=ASUSTeK COMPUTER INC.
uevent=MODALIAS=dmi:bvnAmericanMegatrendsInc.:bvrTP300LA.205:bd11/09/2015:svnASUSTeKCOMPUTERINC.:pnTP300LA:pvr1.0:rvnASUSTeKCOMPUTERINC.:rnTP300LA:rvr1.0:cvnASUSTeKCOMPUTERINC.:ct10:cvr1.0:
```

##### Chuwi Hi10 Pro (HQ64) (cheap atom-based tablet)

```
~ $ (cd /sys/devices/virtual/dmi/id/ ; for f in *; do echo -n "$f="; cat "$f"; done)
bios_date=11/10/2016
bios_vendor=American Megatrends Inc.
bios_version=5.11
board_asset_tag=Hampoo_reserve
board_name=Cherry Trail CR
board_serial=cat: board_serial: Permission denied
board_vendor=Hampoo
board_version=V103
chassis_asset_tag=Default string
chassis_serial=cat: chassis_serial: Permission denied
chassis_type=3
chassis_vendor=Default string
chassis_version=Default string
modalias=dmi:bvnAmericanMegatrendsInc.:bvr5.11:bd11/10/2016:svnCHUWIINNOVATIONANDTECHNOLOGY(SHENZHEN)CO.LTD:pnHi10protablet:pvrHampoo_reserve:rvnHampoo:rnCherryTrailCR:rvrV103:cvnDefaultstring:ct3:cvrDefaultstring:
power=cat: power: Is a directory
product_family=CherryTrail CR -- Hampoo
product_name=Hi10 pro tablet
product_serial=cat: product_serial: Permission denied
product_uuid=cat: product_uuid: Permission denied
product_version=Hampoo_reserve
subsystem=cat: subsystem: Is a directory
sys_vendor=CHUWI INNOVATION AND TECHNOLOGY(SHENZHEN)CO.LTD
uevent=MODALIAS=dmi:bvnAmericanMegatrendsInc.:bvr5.11:bd11/10/2016:svnCHUWIINNOVATIONANDTECHNOLOGY(SHENZHEN)CO.LTD:pnHi10protablet:pvrHampoo_reserve:rvnHampoo:rnCherryTrailCR:rvrV103:cvnDefaultstring:ct3:cvrDefaultstring:
```

##### Custom PC build using a P5N7A-VM motherboard

```
~ $ (cd /sys/devices/virtual/dmi/id/ ; for f in *; do echo -n "$f="; cat "$f"; done)
bios_date=04/29/2010
bios_vendor=American Megatrends Inc.
bios_version=0519   
board_asset_tag=To Be Filled By O.E.M.
board_name=P5N7A-VM
board_serial=cat: board_serial: Permission denied
board_vendor=ASUSTeK Computer INC.
board_version=Rev 1.xx
chassis_asset_tag=Asset-1234567890
chassis_serial=cat: chassis_serial: Permission denied
chassis_type=3
chassis_vendor=Chassis Manufacture
chassis_version=Chassis Version
modalias=dmi:bvnAmericanMegatrendsInc.:bvr0519:bd04/29/2010:svnSystemmanufacturer:pnSystemProductName:pvrSystemVersion:rvnASUSTeKComputerINC.:rnP5N7A-VM:rvrRev1.xx:cvnChassisManufacture:ct3:cvrChassisVersion:
power=cat: power: Is a directory
product_family=To Be Filled By O.E.M.
product_name=System Product Name
product_serial=cat: product_serial: Permission denied
product_uuid=cat: product_uuid: Permission denied
product_version=System Version
subsystem=cat: subsystem: Is a directory
sys_vendor=System manufacturer
uevent=MODALIAS=dmi:bvnAmericanMegatrendsInc.:bvr0519:bd04/29/2010:svnSystemmanufacturer:pnSystemProductName:pvrSystemVersion:rvnASUSTeKComputerINC.:rnP5N7A-VM:rvrRev1.xx:cvnChassisManufacture:ct3:cvrChassisVersion:
```

##### HP Z420 Workstation

```
~ $ (cd /sys/devices/virtual/dmi/id/ ; for f in *; do echo -n "$f="; cat "$f"; done)
bios_date=06/01/2016
bios_vendor=Hewlett-Packard
bios_version=J61 v03.90
board_asset_tag=DUFFMAN
board_name=1589
board_serial=cat: board_serial: Permission denied
board_vendor=Hewlett-Packard
board_version=0.00
chassis_asset_tag=DUFFMAN
chassis_serial=cat: chassis_serial: Permission denied
chassis_type=6
chassis_vendor=Hewlett-Packard
chassis_version=
modalias=dmi:bvnHewlett-Packard:bvrJ61v03.90:bd06/01/2016:svnHewlett-Packard:pnHPZ420Workstation:pvr:rvnHewlett-Packard:rn1589:rvr0.00:cvnHewlett-Packard:ct6:cvr:
power=cat: power: Is a directory
product_family=103C_53335X G=D
product_name=HP Z420 Workstation
product_serial=cat: product_serial: Permission denied
product_uuid=cat: product_uuid: Permission denied
product_version=
subsystem=cat: subsystem: Is a directory
sys_vendor=Hewlett-Packard
uevent=MODALIAS=dmi:bvnHewlett-Packard:bvrJ61v03.90:bd06/01/2016:svnHewlett-Packard:pnHPZ420Workstation:pvr:rvnHewlett-Packard:rn1589:rvr0.00:cvnHewlett-Packard:ct6:cvr:
```

samueldr, 2018-01-16:
There is one big flaw that will need to be adressed: those vendor names aren't awesome!

 * Hewlett-Packard
 * CHUWI INNOVATION AND TECHNOLOGY(SHENZHEN)CO.LTD
 * ASUSTeK COMPUTER INC.
 * ASUSTeK Computer INC.
 * GOOGLE

I wouldn't be surprised if they weren't stable across different products, either from capitalization, spacing or even dropping or adding "inc." and similar suffixes/prefixes. There's already two variants for Asus.

This is one detail we need to check for: how should this be handled? Strip non-letters, then lowercase and then fuzzymatch in some way to an empirically produced list of vendors?

samueldr, 2018-01-16:
Oh, and there are fields in the DMI informations, not show here, that *may* be useful depending on the vendor:

```
$ sudo nix-shell -p dmidecode --run "dmidecode -t 11"
```

DON'T POST THOSE HERE before looking up on google if there are any hits.

For *at least* Asus laptops, these strings define the specific SKU for the model, [as their model name and model numbers vary wildly](https://nixos.wiki/wiki/Hardware/Asus) I was able to find the model number `90NB05Y1-M01340` for my hardware in String 4, and from a google search of dmidecode results, it seems that the String 4 for Asus laptops is a good match to find the exact hardware specs when multiple laptops share the same model name.

yegortimoshenko, 2018-01-16:
Going this route, it would make more sense to put effort into `nixos-generate-config` for specific hardware parts rather than identify devices (and apply per-device profile).

Device profiles are inherently problematic: assumption is made that there is next to no hardware configuration made. That is not actually the case: many components can be replaced (Wi-Fi card from a different vendor, HDD -> SSD, etc.). It's also not an option for custom-built hardware, like gaming rigs, or most desktop systems, even those that have model numbers.

To handle the complexity of identifying hardware better, it's likely that `nixos-generate-config` might have to be rewritten in a more declarative way, at least the hardware identification part.

samueldr, 2018-01-16:
Definitely! Detecting the hardware components in the device is 100% better than hoping the device has that hardware. Just think of the Asus example, they use the same model name (TP300LA) for devices with and without NVIDIA GPU, which could wreak havoc when someone with good intentions add a relevant kernel module.

The approach udev/systemd seems to use with regards to devices (computers, laptops, tablets) is only to apply *quirks* to them. E.g. that macbook model is **known** to need this workaround for *problem*; that sensors mounted on this *model* is **known** to be installed in that orientation.

It might still be relevant to implement the *configuration* of the identified components here though, even if the detection is handled in the configuration generation scripts.

There may be some need for leeway into splitting devices and components. The Chuwi Hi10 Pro device has a Cherry Trail Atom CPU, and is built on a common Cherry Trail platform which has *some quirks*, but those quirks are not specific to that device. In fact, most of the quirks of the GPD 7 with regards to charging, backlight and power use are the same than are present in the Chuwi Hi10 Pro. This would make the "Cherry Trail Platform" a kind of device, but one that may be harder to detect, and might need some *creative license* for the hardware devices to include. (I don't know if this part all made sense.)

> * * *
> 
> ### Glossary
> 
> (Definitions for terms used in this post to reduce confusion)
> 
> #### Component
> 
> A hardware piece used in a *device*. E.g.: GPU, CPU, sensor, etc.
> 
> #### Device
> 
> A *complete* package of components, more often called a computer, laptop, notebook or tablet.
>
> * * *

nixos-discourse, 2020-05-29:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/best-practice-for-enabling-hardware-support/7424/1