NixOS/nixos-hardware #50: Repository should be linked from the NixOS manual

Baughn, 2018-02-01:
At least, assuming it's generally valuable.

I had no idea this existed, and just ran into it by accident. For bonus points, have nixos-generate-config propose the profile addition.

garbas, 2018-02-01:
👍 i think we should - at least - mention it in generated configuration.nix

yegortimoshenko, 2018-02-02:
I also think this should make it into the manual.

> At least, assuming it's generally valuable.

It is! Although, it would be great to see more contributed hardware profiles.

> For bonus points, have `nixos-generate-config` propose the profile addition.

`nixos-generate-config` is non-interactive. Individual hardware part configurations should be 
moved to the script, and eventually (long-term) perhaps there would be no need in this repository. While practical, this approach is [problematic](https://github.com/NixOS/nixos-hardware/issues/49#issuecomment-358067164).

testbird, 2019-06-21:
I'm actually surprised the hardware is not autodetected and corresponding profiles configured automatically, given that nix is said to be a programming language to configure operating systems. Couldn't you try an option like that instead of adding manual instructions?

Moredread, 2019-06-22:
@testbird I think having both would be useful, but I don't know how nixos-generate-config works and if detecting e.g. different laptop models would even work automatically.

samueldr, 2019-06-23:
Using the contents of `/sys/devices/virtual/dmi/id/*` it *should* be possible to match specific models. That is the same information that is used by some quirks in-kernel or in systemd.

I say *should*, because for some OEMs, it seems awfully hard to just put the right information in there. The main issue seems mostly that DMI information is too generic in those cases, I do not know of hardware falsely claiming to be something it isn't in the DMI information.

wmertens, 2019-06-24:
So if we can generate a hash based on the contents of that directory, and it is somewhat stable, we can have the configurations be keyed by that hash. New models just mean more hashes, and if there are two clashing models, we can augment the hash with extra information from elsewhere?

testbird, 2019-06-24:
Why would you narrow things down with hashes, won't globing make more sense to configure some more general things that are to be applied across small version changes and DMI differences, as long as no contrary case is known?
sys_vendor==Somename
product_version==*System X123* (also catches X123v2 for configurations where it makes sense)

wmertens, 2019-06-24:
Well, with a hash it's easy to match, and you can be sure everything works as it did for the other users of the same hardware. By doing matching like you propose, there can be a lot of errors, and I think it would be safer to confirm the entire set of components, and if there's anything even a little bit different, a human has to look at it.

edolstra, 2019-06-24:
The `nixos-hardware` repo is just a bunch of configurations that people may find useful. It should not be included by default. If there are things in there that are unambiguously a good idea, they should be included in the `nixpkgs` repo or upstreamed to e.g. systemd (like missing `udev` rules).

testbird, 2019-06-24:
I think hashes would just make unnecessary computations and things harder to debug.

Also avoid failures: Small things always change in models, even it it's just the manufacturing date, without any relevance and need to change software configurations.

wmertens, 2019-06-24:
I think it would be nice to see a message "your system matches this/these configurations at nixos-hardware, maybe you'd like to look at those". If a hash changes for a later revision of the same model, just add it to the list of hashes for that configuration.

testbird, 2019-06-24:
Upstreaming model identifier matching (e.g. regexp) and logging sounds much better to me than adding user prompts.

yegortimoshenko, 2019-06-26:
@edolstra `nixos-hardware` has grown to be much more formal than just a bunch of configurations, it aims for just supporting all of underlying hardware, and working around any quirks that it might have, without anything user-specific or opinionated on top of it.

For example:
- [Dell XPS 13 profile](https://github.com/NixOS/nixos-hardware/blob/master/dell/xps/13-9380/default.nix)

  Works around battery drain during sleep.

- [Macbook Air 6 profile](https://github.com/NixOS/nixos-hardware/blob/master/apple/macbook-air/6/default.nix)

  Sets power consumption workaround (`acpi_osi=`), imports third-party kernel module for backlight.

- [ThinkPad X280 profile](https://github.com/NixOS/nixos-hardware/blob/master/lenovo/thinkpad/x270/default.nix)

  No workarounds necessary: inherits common ThinkPad configuration, which in turn inherits common laptop configuration. Also inherits common Intel CPU configuration, which sets up microcode updates.

I do believe this is generic enough to be included into Nixpkgs. If there are any rough edges, these could be reconciled during merge.

matthewbauer, 2019-06-26:
:+1: for integrating it into nixos-generate-config.pl (or some successor scripts). I'm not sure if there is an easy way to detect "laptop" in the same way we can detect USB devices though. Usually these quirks fixed in nixos-hardware come from the BIOS itself, although a few are related to USB devices directly.

Mic92, 2019-07-02:
`dmidecode` should be able to detect laptop models.

testbird, 2019-07-02:
Does't the kernel have and provide all the required info already?

E.g, see:
 `for info in /sys/devices/virtual/dmi/id/* ;  do if [ -r $info ] && [ -f $info ]; then printf "%-17s : %-30s\n" "${info##*/}" "$(cat $info)"; fi; done`

wmertens, 2019-07-02:
How about having nixos-generate-config collect the DMI info and provide it
as extra data to the hw config? Then all kinds of hardware-specific
workarounds could look at that data and live in the nixpkgs tree?

testbird, 2019-07-02:
I like that idea. Loading the */sys/devices/virtual/dmi/id/*` data readily available for the modules might even allow for "hardware dependent" package dependencies, or even system package declarations and thus automatic installations.

For example, some packages covering a set of models, and a generic package covering common cpu quirk stuff.

davidak, 2019-08-14:
I like the idea of `nixos-generate-config` collecting data about the hardware and put them in `hardware-configuration.nix`. @danbst mentioned it in https://github.com/NixOS/nixos-hardware/pull/101#issuecomment-465739272 already.

But i would use a tool or library to detect the hardware model that can do that robustly rather than developing something ourselves. [neofetch](https://github.com/dylanaraps/neofetch) does that for example. It's just a bash script, so no extra dependencies needed and it's 278K big.

```
$ neofetch model
model: 20J4000XGE ThinkPad L470

$ neofetch cpu
/run/current-system/sw/bin/neofetch: line 1830: /sys/devices/system/cpu/cpu0/cpufreq/bios_limit: No such file or directory
/run/current-system/sw/bin/neofetch: line 1831: /sys/devices/system/cpu/cpu0/cpufreq/bios_limit: No such file or directory
cpu: Intel i7-7500U (4) @ 3.500GHz

$ neofetch gpu
gpu: Intel HD Graphics 620 
gpu: AMD ATI Radeon HD 8670A/8670M/8690M / R5 M330 / M430 / R7 M520
```

Maybe it is even something that could be part of systemd?

Currently any program does that on it's own and that's a huge code duplication.

For example the About panel of Pantheon Desktop settings shows CPU and GPU. It would be great to see the hardware Model there too, like on macOS! And when systemd can provide that, all programs can just use it (and have a dependency on systemd :D). So maybe a library as part of systemd that can be used independently.

[image]

[image]

zimbatm, 2020-06-22:
related: https://github.com/NixOS/rfcs/pull/70

Mic92, 2021-06-13:
nixos-generate-config advertises this repo. I think this is good enough for now.

closed 2021-06-13 by Mic92