NixOS/nixos-hardware #74: Enable intel-hybrid-driver on some Intel architectures

yegortimoshenko, 2018-08-24:
intel-hybrid-driver adds accelerated VP8/VP9 decoding on systems that normally don't have it, such as Bay Trail devices: https://github.com/01org/intel-hybrid-driver

colemickens, 2018-12-11:
@yegortimoshenko This is interesting to learn about, thanks!

Does this entail more than just enabling the option added in this commit: https://github.com/NixOS/nixpkgs/pull/42673/commits/d2935b93e6a36973c698d5340b0d3e174721e192 ?

colemickens, 2018-12-11:
I added some docs to the Wiki, in the meantime: https://nixos.wiki/wiki/Accelerated_Video_Playback

yegortimoshenko, 2018-12-11:
Oh, didn't see that!

I think just enabling the option will do, here: https://github.com/NixOS/nixos-hardware/blob/master/common/cpu/intel/default.nix#L9

colemickens, 2019-01-30:
```
    nixpkgs.config.packageOverrides = pkgs: {
      vaapiIntel = pkgs.vaapiIntel.override { enableHybridCodec = true; };
    };
```

is what I've been using. Not sure if it's working or not?

Thoughts? I can send a PR?

yegortimoshenko, 2019-01-30:
Looks good!

closed 2024-12-15 by lukateras