NixOS/nixos-hardware #87: dell xps: consider enabling AC power supply "fix"

colemickens, 2018-12-08:
I'm not fully sure of all of the implications, but the last cause of problems with CPU scaling on my XPS 13 9370 was tracked down to weird combinations of my USB Type C Hub and various Type C power supplies that I have around.

Using `msr-tools` and executing `sudo wrmsr -a 0x1FC 2` "fixes" this problem.

I don't fully understand the implications of this, so I'm not going to send a PR right away, but I thought I'd offer it for consideration (and as a potential warning to others in the future).

Also, in the interest of potentially helping other XPS users, the Nov 2018 firmware update has been a big improvement to performance and consistency.

colemickens, 2018-12-08:
@yegortimoshenko yeah, per my note at the bottom, I did just update everything that I could get to from Linux and Windows (BIOS, LCD, SSD, the works). However, as you found, this is still useful with non-OEM powersupplies (or it seems, OEM powersupplies when used through some types of USB Hubs).

My hesitation is that Dell might have a legitimate (safety?) reason for shipping with this default configuration.

Moredread, 2018-12-08:
@colemickens how did you notice the problem? Where did you find the fix?

colemickens, 2018-12-08:
@Moredread I noticed the problem because I recently added my CPU Frequency to my i3bar and observed that when I utilized power through the USB Hub, the CPU Frequency would immediately scale to 10% (aka, 400Mhz on my processor). Turns out, that's rather noticeable...

Found the fix by googling "Dell" "AC" "cpu scaling", or something along those lines. There were strong indicators that the combination of hardware (cpu, power, hub, Dell HW) were interacting poorly.

colemickens, 2019-02-14:
https://superuser.com/a/1357098

Moredread, 2019-02-15:
I'm using https://github.com/erpalma/throttled, packaged in my nur repo at https://github.com/Moredread/nur-packages/blob/master/pkgs/lenovo-throttling-fix/default.nix and https://github.com/Moredread/nur-packages/blob/master/modules/lenovo-throttling-fix/default.nix

I didn't find that msr address in the source, but maybe it already is handled. For temperature throttling the fix needs to be applied periodically or at least after each sleep wake up, which that tool handles.

Moredread, 2019-02-15:
If it look interesting to you I might try to get it added before 19.03

colemickens, 2019-02-15:
@Moredread As I understand it, the most important fix performed by `throttled` is already incorporated by this repo and is already included in the XPS 13 9370 configuration already. See here and compare: https://github.com/NixOS/nixos-hardware/blob/master/common/pc/laptop/cpu-throttling-bug.nix#L14 with: https://github.com/erpalma/throttled/blob/master/lenovo_fix.py#L461

However, it looks that `thermald` more stuff. Do you know conclusively if the other things it does are necessary on the XPS 13?

Either way, this particular thread is for a different, somewhat related issue that corresponds with certain combinations of chargers/docks and certain Dell laptops and is independent from temperature throttling. If we think thermald is useful beyond just resetting the temperature trigger threshold, maybe we should have a new PR/Issue to discuss it.

Mic92, 2021-06-13:
We have both throttled and thermald. 13-9370 profile also enables it.

closed 2021-06-13 by Mic92

colemickens, 2021-06-14:
(This was about a different MSR controlled behavior than the one that those two tools handle. But it's probably a safety mechanism and I no longer have that laptop, so we can leave this closed.)