NixOS/nixos-hardware #90: Listing derivations (nix-env -qa) does not work

zarelit, 2018-12-30:
When I try to list available packages with `nix-env -qaP` the evaluation breaks in `nixos-hardware/default.nix` and I don't know what's wrong. The following logs are taken as root but it behaves the same with an unprivileged user.

**nix-env -qaP --show-trace**
```
[root@tsundoku:/home/zarel]# nix-env -qaP --show-trace
error: while evaluating 'buildProfile' at /nix/store/rkb722vhq7xkhid0nbnl55i2r59csd77-nixos-hardware/nixos-hardware/default.nix:18:18, called from /nix/store/rkb722vhq7xkhid0nbnl55i2r59csd77-nixos-hardware/nixos-hardware/default.nix:51:20:
while evaluating the attribute 'system' at /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/default.nix:33:3:
while evaluating the attribute 'config.system.build.toplevel' at /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/modules/system/activation/top-level.nix:273:5:
while evaluating 'foldr' at /nix/var/nix/profiles/per-user/root/channels/nixos/lib/lists.nix:39:20, called from /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/modules/system/activation/top-level.nix:137:12:
while evaluating 'fold'' at /nix/var/nix/profiles/per-user/root/channels/nixos/lib/lists.nix:42:15, called from /nix/var/nix/profiles/per-user/root/channels/nixos/lib/lists.nix:46:8:
while evaluating anonymous function at /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/modules/system/activation/top-level.nix:137:52, called from /nix/var/nix/profiles/per-user/root/channels/nixos/lib/lists.nix:45:14:
while evaluating anonymous function at /nix/store/vs30cml2ma72nd6vavfzc7ljrqw3p356-nixos-18.09.1819.76aafbf4bf4/nixos/pkgs/build-support/replace-dependency.nix:20:1, called from /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/modules/system/activation/top-level.nix:138:7:
cannot import '/nix/store/parl2dsdfc22mzdqcfi9ic60r9cgp4k4-references.nix', since path '/nix/store/k0my68piibb4180drdqcpvxnla2fd76w-references.nix.drv' is not valid, at /nix/store/vs30cml2ma72nd6vavfzc7ljrqw3p356-nixos-18.09.1819.76aafbf4bf4/nixos/pkgs/build-support/replace-dependency.nix:26:16
```

**nix-info**

```
[root@tsundoku:/home/zarel]# nix-info
system: "x86_64-linux", multi-user?: yes, version: nix-env (Nix) 2.1.3, channels(root): "nixos-hardware, nixos-18.09.1819.76aafbf4bf4", channels(zarel): "nixpkgs-19.03pre161900.61c3169a0e1", nixpkgs: /nix/var/nix/profiles/per-user/root/channels/nixos
```

**nix-channel --list**

```
[root@tsundoku:/home/zarel]# nix-channel --list
nixos https://nixos.org/channels/nixos-18.09
nixos-hardware https://github.com/NixOS/nixos-hardware/archive/master.tar.gz
```

The nixos-rebuild operations work as expected.

vyorkin, 2019-01-01:
got same error,
**nix-env -qaP --show-trace**:
```
error: while evaluating 'buildProfile' at /nix/store/dy6jxh9xyyddg22z3d81cjsk0c4hv4wm-nixos-hardware/nixos-hardware/default.nix:18:18, called from /nix/store/dy6jxh9xyyddg22z3d81cjsk0c4hv4wm-nixos-hardware/nixos-hardware/default.nix:51:20:
while evaluating the attribute 'system' at /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/default.nix:33:3:
while evaluating the attribute 'config.system.build.toplevel' at /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/modules/system/activation/top-level.nix:276:5:
while evaluating 'foldr' at /nix/var/nix/profiles/per-user/root/channels/nixos/lib/lists.nix:39:20, called from /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/modules/system/activation/top-level.nix:140:12:
while evaluating 'fold'' at /nix/var/nix/profiles/per-user/root/channels/nixos/lib/lists.nix:42:15, called from /nix/var/nix/profiles/per-user/root/channels/nixos/lib/lists.nix:46:8:
while evaluating anonymous function at /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/modules/system/activation/top-level.nix:140:52, called from /nix/var/nix/profiles/per-user/root/channels/nixos/lib/lists.nix:45:14:
while evaluating anonymous function at /nix/var/nix/profiles/per-user/root/channels/nixos/pkgs/build-support/replace-dependency.nix:20:1, called from /nix/var/nix/profiles/per-user/root/channels/nixos/nixos/modules/system/activation/top-level.nix:141:7:
cannot import '/nix/store/li9f2kcra8gy76v78w90gg78q861w27n-references.nix', since path '/nix/store/7hh77nsis8hbx0451dw0jhcalr856dn4-references.nix.drv' is not valid, at /nix/var/nix/profiles/per-user/root/channels/nixos/pkgs/build-support/replace-dependency.nix:26:16
```
**nix-info**:
```
system: "x86_64-linux", multi-user?: yes, version: nix-env (Nix) 2.1.3, channels(root): "nixos-hardware, nixos-19.03pre164715.201d739b0ff", nixpkgs: /nix/var/nix/profiles/per-user/root/channels/nixos
```
**nix-channel --list**:
```
[root@silence:/home/vyorkin]$ nix-channel --list
nixos https://nixos.org/channels/nixos-unstable
nixos-hardware https://github.com/NixOS/nixos-hardware/archive/master.tar.gz
```

Mic92, 2019-01-04:
Looks like import from derivation @yegortimoshenko

Mic92, 2019-01-04:
Since we are not really needing those for the channel, the quick fix would be to move `default.nix` to `release.nix`.

closed 2019-01-09 by Mic92

zarelit, 2019-01-09:
@Mic92  @yegortimoshenko  Cool! Thank you for all your efforts on nixos-hardware

Mic92, 2019-01-18:
I am not quiet sure where this error is coming form. I don't say any file that takes a `pkgs` argument. Does the following pr fixes your https://github.com/NixOS/nixos-hardware/issues/90#issuecomment-452679115?

Mic92, 2019-01-18:
use the following for testing:

```console
$ nix-channel --add https://github.com/NixOS/nixos-hardware/archive/nix-env-part2.tar.gz nixos-hardware
```

zarelit, 2019-01-18:
Updated the channel after the merge of https://github.com/NixOS/nixos-hardware/pull/95 and now it works, thank you!