NixOS/nixos-hardware #94: Using the nixos-channel makes nix-env --upgrade to fail

fatpat, 2019-01-10:
Hi,

I have an issue when having the nixos-hardware in my channels list: `nix-env --upgrade` is broken

```
[fat@fat-openio]$ sudo -i nix-channel --list
nixos https://nixos.org/channels/nixos-18.09
nixos-hardware https://github.com/NixOS/nixos-hardware/archive/master.tar.gz
```

```
[fat@fat-openio]$ nix-channel --list
nixos-unstable https://nixos.org/channels/nixos-unstable
```

```
[fat@fat-openio]$ sudo -i nix-channel --update
unpacking channels...

[fat@fat-openio]$ nix-channel --update
unpacking channels...
```

```
fat@fat-openio]$ nix-env --verbose -u firefox
evaluating file '/nix/store/rffcxk0l94lc96yl07r26sdnfql6x0h4-nix-2.1.3/share/nix/corepkgs/derivation.nix'
evaluating file '/nix/store/d31qkkqp4s78fs9wixwb72rqyv8dq89y-env-manifest.nix'
evaluating file '/home/fat/.nix-defexpr/channels_root/nixos-hardware/supermicro/a1sri-2758f/default.nix'
error: cannot auto-call a function that has an argument without a default value ('pkgs')
```

If I remove the nixos-hardware channel, nix-env works great. 

Has someone an idea for fixing or helping debugging the issue ?

Thanks

zarelit, 2019-01-18:
I had this issue but it looks like it has been fixed after the merge of https://github.com/NixOS/nixos-hardware/pull/95
@fatpat  could you `sudo nix-channel --update` and let us know if it has been fixed?

Mic92, 2019-01-18:
Should be the case.

closed 2019-01-18 by Mic92

fatpat, 2019-01-18:
it seems to be fixed ... I have other issues with nix-env so I cannot really confirm