NixOS/nixos-hardware #96: XPS 13 (9370): Additional i915 flags to set?

colemickens, 2019-01-24:
Please note, I don't have hard data for any of this, I can't say if these things are guaranteed improvements -- its more that I trust [the Arch Wiki's XPS13 page](https://wiki.archlinux.org/index.php/Dell_XPS_13_(9360)#Video) and have been running these settings for sometime.

In my kernel params, I have some extras that aren't included via the xps-13-9370 file in this repository.

```
        "i915.modeset=1"       # entirely absent in nixos-hardware
        "i915.enable_guc=2" # entirely absent in nixos-hardware
        "i915.enable_gvt=1" # entirely absent in nixos-hardware
        "i915.enable_psr=1" # entirely absent in nixos-hardware
        "i915.fastboot=1"     # entirely absent in nixos-hardware

        "i915.enable_fbc=1" # set to 2 in nixos-hardware
```

I think I checked everywhere those could be getting set, but I could be incorrect about all of those being missing.

Thoughts from other XPS 13 users?

colemickens, 2019-01-24:
Note that the wiki page says `i915.enable_guc=3` but if I recall correctly, it failed to boot, or failed to drive the screen properly when I did that, hence why I have it to set to `2`. I don't recall the meaning of 2 vs 3.

colemickens, 2019-01-24:
Hm, looks like GVT (gpu virtualization) is incompatible with GuC loading, which might explain some of the issues I (was|am) having. I'll do some additional testing and reply back with some more concrete information. https://wiki.archlinux.org/index.php/intel_graphics#Enable_GuC_/_HuC_firmware_loading

grahamc, 2019-01-24:
```
i915_param_named(modeset, int, 0400,
        "Use kernel modesetting [KMS] (0=disable, "
        "1=on, -1=force vga console preference [default])");
```

I don't think guc=3 has a meaning:

```
i915_param_named_unsafe(enable_guc, int, 0400,
        "Enable GuC load for GuC submission and/or HuC load. "
        "Required functionality can be selected using bitmask values. "
        "(-1=auto, 0=disable [default], 1=GuC submission, 2=HuC load)");
```

```
i915_param_named(enable_gvt, bool, 0400,
        "Enable support for Intel GVT-g graphics virtualization host support(default:false)");
```

uh?: 

```
i915_param_named_unsafe(enable_psr, int, 0600,
        "Enable PSR "
        "(0=disabled, 1=enabled) "
        "Default: -1 (use per-chip default)");
```

```
i915_param_named(fastboot, bool, 0600,
        "Try to skip unnecessary mode sets at boot time (default: false)");

```

```
i915_param_named_unsafe(enable_fbc, int, 0600,
        "Enable frame buffer compression for power savings "
        "(default: -1 (use per-chip default))");
```

colemickens, 2019-01-28:
Thanks @grahamc, I see now that reading the source is easier than trying to find public docs for the flags... heh.

* KMS works on XPS 13 9370, so it's safe to enable if it's not already.
* GUC is a bitmask, 3 is HUC_load+GUC_submission (I didn't actually realize this until re-reading the text you quoted in fact)
* Fastboot is something intel is looking to enable by default anyway, and works on (my) XPS 13 9370.
* the rest are powersaving features that all work on (my) XPS 13 9370.

Moredread, 2019-01-29:
Thanks for looking into it. I'll try those changes on my laptop

Moredread, 2019-02-02:
I've just tried those params and the XPS 13 still works :p I haven't tested the power usage though.

Moredread, 2019-02-08:
I had problems with freezing alacritty windows (not refreshing anymore, keeping parts of the previous screenspace when switching workspace in i3) the last view days but am not sure how to reproduce this.

I'll try again with and without the flags to see if it makes a difference. :/

colemickens, 2019-02-08:
Interesting! I'll disable my changes for the next week and observe as well.

Moredread, 2019-02-16:
I don't see the issue anymore, with only

```
        "i915.modeset=1"       # entirely absent in nixos-hardware
        "i915.fastboot=1"     # entirely absent in nixos-hardware
```

I'll try to bisect the options to see what exactly causes this. :/

Moredread, 2019-04-05:
I have experimented setting only certain flags over the last months, but I it never went totally away. I have deactivated them for about 2 weeks now, and it still happens. :/

It seems likely it is a general issue for my system after sleep. The only thing I haven't tried yet is disabling "mem_sleep_default=deep".

@colemickens Did you notice anything?

sameer, 2019-09-04:
> Note that the wiki page says `i915.enable_guc=3` but if I recall correctly, it failed to boot, or failed to drive the screen properly when I did that, hence why I have it to set to `2`. I don't recall the meaning of 2 vs 3.

For future readers: 3 is a bitmask value of 1 & 2