NixOS/nixos-hardware #98: Lenovo X1 Battery Tresholds set

eyJhb, 2019-02-14:
In the hardware configuration file for Lenovo Thinkpad x1 6th generation, there are some configuration regarding charging thresholds.
As these are personal preference, shouldn't these be removed? 

https://github.com/NixOS/nixos-hardware/blob/master/lenovo/thinkpad/x1/6th-gen/default.nix#L18-L19

yegortimoshenko, 2019-02-14:
I think they should be. cc @azazel75, is that OK with you?

azazel75, 2019-02-15:
They are a *preference*, yes, although not personal: it's more or less the suggested setting on TLP wiki here: https://linrunner.de/en/tlp/docs/tlp-faq.html#battery. In my experience with these systems it's really better to have battery life saving setting, from the start, or you'll find yourself thinking about it after an year  when you realize to have a battery fucked up even if  you kept the notebook plugged  in most of the time. 
It would be nice to have the module extended to be able to set an mkDefault on those settings. So in the end to summarize:
- Are those settings "personal"? Yes, maybe.
- Do we do a better service to the generic user by leaving the default charge to 100%? I don't think so

Maybe a solution in the middle would be to set all the `extraConfig` setting with an mkDefault so that it can be overridden easily?

eyJhb, 2019-02-15:
@azazel75 but there is no other distro, as far as I know, that does this. And this is only for one laptop, and not the whole Thinkpad series. 
Besides, it isn't even enabled by default, so it is only enabled when a user enables TLP (which should indicated, that they most know something about what they are doing).

If it is going to be enabled at all (which I still think it shouldn't), then it should use the "default" Lenovo values, which are 96% and 100% (I think? - [here](https://www.lenardgunda.com/thinkpad-battery-charging-thresholds/))

eyJhb, 2019-02-19:
Well, unless @azazel75 makes a very good point, I will create a PR to remove this.
This is not something that is done for any other laptop configuration, and it is not our decision.

If it were to be done, it should be done higher up, and not for a single laptop only.

eyJhb, 2019-02-19:
Created PR, please merge :)

PR #100 (2019-02-19): lenovo-x1: removed battery TLP threshholds
Fixes #98 , as it should not be set.
closed 2019-02-19 by yegortimoshenko