NixOS/nixos-hardware #99: Enable intel-media-driver?

colemickens, 2019-02-18:
cc: @yegortimoshenko 

Today I leaned that there is another driver from Intel that works with libva for accelerated video.

IF you add `intel-media-driver` to opengl packages, you can do this
`LIBVA_DRIVER_NAME=iHD vainfo` and see that a different driver is being used.

I don't really understand the pro/cons or any differences between using i965 and iHD, but I wanted to mention it since it doesn't seem widely documented/known.

I documented this on the related wiki page: https://nixos.wiki/wiki/Accelerated_Video_Playback

colemickens, 2019-03-22:
I don't see any reason to not include this. AFAICT it's always going to require the user to opt in for it to be prioritized over i915. I'll send a PR.

asymmetric, 2019-10-25:
@colemickens did you eventually create the PR?

PR #131 (2020-05-07): intel: add intel-media-driver
Provides hardware accelerated video playback from Broadwell onwards.

Older boards (e.g. Sandy Bridge) are not supported. I'm not sure if they should specifically be excluded.

More info:

* https://github.com/intel/media-driver#supported-platforms
* https://wiki.archlinux.org/index.php/Hardware_video_acceleration#Intel

Closes #99.
closed 2020-05-07 by Mic92