NixOS/nixos-hardware #105: Support for Lenovo X1 Extreme

craisp, 2019-04-30:
Hi there

I've been using the Lenovo X1 Extreme with NixOS and ran into a lot of problems regarding the hardware support.
Especially the hybrid graphics (Intel and Nvidia) including the HDMI outlet and the webcam don't seem to be working.

The ArchLinux people seem to have figured out most of the problems:
https://wiki.archlinux.org/index.php/Lenovo_ThinkPad_X1_Extreme

I would love to continue using NixOS on this device.
Are there any plans to extend this overlay for the X1 Extreme?

Thanks a million

Mic92, 2019-04-30:
This depends on somebody owning the hardware and contributing their configuration. If you find some of the options from the wiki working for you and you can come up with a working NixOS configuration, feel free to add it.

zfnmxt, 2019-06-09:
I have the same laptop. My webcam just works with no configuration, so I can't help you there.

As for the HDMI output--it's wired to the Nvidia card, not the Intel card. So, to have any output, you need the Nvidia card to be active. This is discussed in the Arch wiki entry you posted above. I went the Bumblebee route to activate the card because I wanted the output to work under the existing X11 server. 

Here's a snippet from my `configuration.nix` that includes the relevant code to get Bumblebee working on the X1E:

```nix
  services.xserver = {
    videoDrivers = [ "intel" ];
    deviceSection = ''BusID "PCI:0:2:0"'';
  };

  hardware.bumblebee = {
    enable = true;
    connectDisplay = true;
  };

  nixpkgs.config.packageOverrides = pkgs: rec {
    bumblebee = pkgs.bumblebee.override {
      extraNvidiaDeviceOptions = ''
        Option "ProbeAllGpus" "false"
        Option "AllowEmptyInitialConfiguration"
      EndSection

      Section "Screen"
        Identifier "Default Screen"
        Device "DiscreteNvidia"
      '';
    };
  };

```

After this, you just issue a simple `primusrun intel-virtual-output -f` and your HDMI output should show up as a virtual screen `xrandr`.

ar1a, 2020-04-24:
Is this closed in #123?

Mic92, 2020-04-28:
Looks like.

closed 2020-04-28 by Mic92