NixOS/nixos-hardware #108: Use `fdprint-thinkpad`

Synthetica9, 2019-05-23:
https://github.com/NixOS/nixos-hardware/blob/59c97be4d0c9e859cb8e3d214707601f213a6914/lenovo/thinkpad/default.nix#L9

Shouldn't the `services.fprintd.package` also be set to `fdprint-thinkpad`? (Introduced in https://github.com/NixOS/nixpkgs/pull/61546)

CC @Cizra

cf6b88f, 2019-05-23:
Possibly. I've only tested it on Lenovo P50, which might be different from other ThinkPads.

It's `fprintd-thinkpad`, not `fdprint-thinkpad`.

rnhmjoj, 2020-03-03:
This package is so undocumented that I thought nixpkgs could have been compromised.
I can't find the software homepage by a search of `libfprint-thinkpad` online: I had to open the source of the package to figure out it's https://github.com/3v1n0/libfprint. I think this is because the name was made up by a nixpkgs [contributor](https://github.com/NixOS/nixpkgs/pull/61546), as the software as no official name.
The README.md doesn't exactly explain what it is either, I guess it's a fork to support a specific reader (Validity 138a:0090?), who maintains it and whether the are affiliated with the devs of libfprint (apparently they are).

Anyway, it should be used only on recent thinkpad.