NixOS/nixos-hardware #109: Source channel declaratively?

Spass1966, 2019-05-26:
I'd like to get up and running on a new system without remembering to add the channel beforehand. Is there a way to do this with, for example an overlay?

kalbasit, 2019-05-26:
You can *pin* the channel in your `/etc/nixos/configuration.nix`. See [FAQ/Pinning Nixpkgs](https://nixos.wiki/wiki/FAQ/Pinning_Nixpkgs).

```nix
let
  nixos-hardware = builtins.fetchTarball {
    url = "https://github.com/NixOS/nixos-hardware/archive/59c97be4d0c9e859cb8e3d214707601f213a6914.tar.gz";
    sha256 = "1mmqj5rsgp8p3x754a26fn6wnkrllrybqs7v9yl6rhqf1dbaawak";
  };
in {
    imports = [
        "${nixos-hardware}/whatever-you-need-here"
    ];
}
```

zarelit, 2019-05-27:
In addition to what @kalbasit wrote, that's what I suggest too, I'd add that you can also not pin a specific commit by setting by using

```nix
  nixos-hardware = builtins.fetchTarball {
    url = "https://github.com/NixOS/nixos-hardware/archive/master.tar.gz";
  };
```
pro: will follow the master branch
con: bad for reproducibility, nixos-rebuild will fetch the whole repo at almost every rebuild

Spass1966, 2019-09-02:
This ended up being my final declaration and it doesn't return any errors:

```
let
  url = "https://github.com/colemickens/nixpkgs-wayland/archive/master.tar.gz";
  waylandOverlay = (import (builtins.fetchTarball url));
  nixos-hardware = builtins.fetchTarball {
    url = "https://github.com/NixOS/nixos-hardware/archive/master.tar.gz";
  };
in

{
  imports =
    [
      # Include hardware quirks
      "${nixos-hardware}/purism/librem/13v3"
    ];
```

closed 2019-09-02 by Spass1966