NixOS/nixos-hardware #121: XPS 13 9370 Battery Drain

colemickens, 2019-09-16:
Does anyone know how to check if deep sleep is properly being forced on the XPS 9370? I'm seeing large battery drains while it's suspended again, and it resumes from suspend faster than I remember when I first enabled this "Fix".

Thanks.

Mic92, 2019-09-18:
It should be logged to dmesg:

```
sudo dmesg | grep ACPI
[20932.542014] ACPI: EC: interrupt blocked
[20932.566863] ACPI: Preparing to enter system sleep state S3
[20932.569453] ACPI: EC: event blocked
[20932.569453] ACPI: EC: EC stopped
[20932.578193] ACPI: Low-level resume complete
[20932.578229] ACPI: EC: EC started
[20932.586689] ACPI: Waking up from system sleep state S3
[20932.592312] ACPI: EC: interrupt unblocked
[20932.620285] ACPI: EC: event unblocked
```

matthewbauer, 2019-09-18:
```
cat /sys/power/mem_sleep
```

should also tell you what linux is using

Mic92, 2021-06-13:
Should we keep this issue open?

closed 2021-06-14 by colemickens