NixOS/nixos-hardware #130: Please provide declarative solution for providing channels

YellowOnion, 2019-11-07:
`nixos-rebuild switch` fails after first boot.

It's quite ridiculous that something mentioned during the installation phase of nixos doesn't even survive the installation phase...

jonringer, 2019-11-07:
you may need to do `boot` then reboot. I believe if certain aspects of systemd get upgraded, then the upgrade will fail.

sheijk, 2021-03-28:
Isn't the issue that the channel needs to be added after installation? Or when are these supposed commands from the readme to be executed?
```
$ sudo nix-channel --add https://github.com/NixOS/nixos-hardware/archive/master.tar.gz nixos-hardware
$ sudo nix-channel --update
```

Mic92, 2021-03-29:
Fixed in https://github.com/NixOS/nixos-hardware/commit/eb070c346bfcc1667aa41200cbd23506b76c89bd

closed 2021-03-29 by Mic92