NixOS/nixos-hardware #133: Dell XPS 13 is now throttled to 95C instead of 97C

colemickens, 2019-11-21:
I think this is related to move to "throttled" to fix the intel throttling issue on various laptops.

While trying to use `undervolt` last night, I realized that something is still overriding the temperature target. Watch:

```bash
[cole@xeep:~/code/nixcfg]$ sudo undervolt --temp 97

[cole@xeep:~/code/nixcfg]$ sudo undervolt --read
temperature target: -3 (97C)
core: -99.61 mV
gpu: -75.2 mV
cache: -99.61 mV
uncore: -99.61 mV
analogio: -99.61 mV

# wait a second

[cole@xeep:~/code/nixcfg]$ sudo undervolt --read
temperature target: -5 (95C)
core: -99.61 mV
gpu: -75.2 mV
cache: -99.61 mV
uncore: -99.61 mV
analogio: -99.61 mV
```

And yet:
```
● lenovo_fix.service - Stop Intel throttling
   Loaded: loaded (/nix/store/ra5wj7nzkgyxgj5w22fbdjnyxscs9a3w-throttled-0.6/lib/systemd/system/lenovo_fix.service; enabled; vendor preset: enabled)
  Drop-In: /nix/store/ys52kwpvs7b3psrllinzlk0axrckhcs7-system-units/lenovo_fix.service.d
           └─overrides.conf
   Active: active (running) since Thu 2019-11-21 04:06:59 PST; 11h ago
 Main PID: 910 (.lenovo_fix.py-)
       IP: 0B in, 0B out
    Tasks: 2
   Memory: 34.5M
      CPU: 3.425s
   CGroup: /system.slice/lenovo_fix.service
           └─910 /nix/store/gpnm7i19lpj8p43mjrdw03d0hjalmskl-python3-3.7.5/bin/python3.7 /nix/store/ra5wj7nzkgyxgj5w22fbdjnyxscs9a3w-throttled-0.6/bin/.lenovo_fix.py-wrapped

Nov 21 04:06:59 xeep systemd[1]: Started Stop Intel throttling.
Nov 21 04:06:59 xeep lenovo_fix.py[910]: [I] Detected CPU architecture: Intel Kaby Lake (R)
Nov 21 04:06:59 xeep lenovo_fix.py[910]: [I] Loading config file.
Nov 21 04:06:59 xeep lenovo_fix.py[910]: [I] Starting main loop.

```

colemickens, 2019-11-21:
I guess it looks like `throttled` can do undervolting as well. Maybe I should only use one or the other.

It also appears that throttled might default to a trip target of 95C, which would explain the behavior I'm seeing.

colemickens, 2019-11-22:
Further, throttled's docs say that it might not be okay to run with thermald at the same time.

Moredread, 2022-06-21:
This might be relevant to the 9370 config, as it enables thermald and throttled.

Moredread, 2022-06-21:
https://github.com/NixOS/nixos-hardware/pull/127 was the PR for that