NixOS/nixos-hardware #140: Way to automatically detect device and apply correct fixes

mkg20001, 2020-02-06:
Don't know if it's possible but there should be a way to do something like `<nixos-hardware/auto-detect>` which would automatically include the right config for the device, which could enable including the right config in an automated fashion.

Otherwise a tool that just returns the right path or nothing if no hardware-quirk conf is needed would be cool

Needing this for scripting purposes

mkg20001, 2020-02-06:
When doing `inxi -v7` it spits out, among other things:
```
Machine:
  Type: Laptop System: LENOVO product: 4334 v: Lenovo G570 
```

Based on those (wherever it got those from) the device detection could work

mkg20001, 2020-02-06:
Seems to be /sys/class/dmi/id/product_*

```
$ tail /sys/class/dmi/id/product_*
==> /sys/class/dmi/id/product_family <==
IDEAPAD

==> /sys/class/dmi/id/product_name <==
4334                            
tail: '/sys/class/dmi/id/product_serial' kann nicht zum Lesen geöffnet werden: Permission denied

==> /sys/class/dmi/id/product_sku <==
HuronRiver_CRB
tail: '/sys/class/dmi/id/product_uuid' kann nicht zum Lesen geöffnet werden: Permission denied

==> /sys/class/dmi/id/product_version <==
Lenovo G570                     
```

samueldr, 2020-02-06:
(See #49, with similar goals.)

Nix itself will not be able to take a peek at the system, especially given the purity focus flakes is having. This, though, could happen in a "nixos-hardware-detect" kind of script.

nixos-discourse, 2020-05-29:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/best-practice-for-enabling-hardware-support/7424/1

tobiasBora, 2020-07-01:
Is there any news on that? I think it could be very interesting to have at least a script that proposes some configuration for the hardware. For example, if I have this graphics card, then I should enable this module... Ubuntu was doing something similar at some point, not sure if we can get inspired by that. But at least a minimal script that handles some basic configuration would be nice.

I just discovered [nixos-hardware](https://github.com/NixOS/nixos-hardware), and I think it's a great project, but I doubt that all computers will enter in that table (mine is not for example), so a more generic solution could be nice. For example it seems that intel usually uses [this configuration](https://github.com/NixOS/nixos-hardware/blob/master/common/cpu/intel/default.nix), does that mean that I should use that as soon as I have a Intel computer? If there are some command that can be run to check if I need `i915` kernel module, then it could be nice to pack it in a script.

jonringer, 2020-07-01:
NixOS has a way to do this through `nixos-generate-config`. There's further discussion here: https://github.com/NixOS/rfcs/pull/70