NixOS/nixos-hardware #160: Intel GuC/HuC support

asymmetric, 2020-05-07:
Intel CPUs from Skylake onwards have a GPU microcontroller that can be used to offload certain tasks to the hardware.

More info:

- https://01.org/linuxgraphics/downloads/firmware
- https://www.kernel.org/doc/html/latest/gpu/i915.html#guc
- https://wiki.gentoo.org/wiki/Intel#Feature_support
- https://wiki.archlinux.org/index.php/Intel_graphics#Enable_GuC_/_HuC_firmware_loading

asymmetric, 2020-05-07:
Worth [keeping in mind](https://mpv.io/manual/stable/#options-hwdec):

> In general, it's very strongly advised to avoid hardware decoding unless **absolutely** necessary, i.e. if your CPU is insufficient to decode the file in questions. If you run into any weird decoding issues, frame glitches or discoloration, and you have --hwdec turned on, the first thing you should try is disabling it.

Mic92, 2021-07-25:
Any updates on this?

mainrs, 2022-08-17:
Is there any guidance/examples for how to implement this properly? I am fairly new to NixOS.

wowpetr, 2022-12-03:
You can just add the kernel parameter to your nixos configuration:
```
boot.kernelParams = [ "i915.enable_guc=3" ];
```

queeup, 2023-09-01:
Well, firmware version mismatch.
```bash
$ sudo cat /sys/kernel/debug/dri/0/gt/uc/guc_info
GuC firmware: i915/kbl_guc_70.1.1.bin
	status: MISSING
	version: wanted 70.1.0, found 0.0.0
	uCode: 0 bytes
	RSA: 0 bytes

GuC status 0x00000001:
	Bootrom status = 0x0
	uKernel status = 0x0
	MIA Core status = 0x0

Scratch registers:
	 0: 	0x0
	 1: 	0x0
	 2: 	0x0
	 3: 	0x0
	 4: 	0x0
	 5: 	0x0
	 6: 	0x0
	 7: 	0x0
	 8: 	0x0
	 9: 	0x0
	10: 	0x0
	11: 	0x0
	12: 	0x0
	13: 	0x0
	14: 	0x0
	15: 	0x0

GuC log relay not created
```

Moe1369, 2024-12-17:
@queeup 
Exact same issue. Is there any workaround for that?

Moe1369, 2024-12-17:
I somewhat fixed it. By manually copying the ehl_guc_70.1.1.bin and huc bin to /lib/firmware/i915. 
Important to note that the kernel parameter i915.guc_firmware_path=${pkgs.linux-firmware}/lib/firmware/i915 does NOT work.