NixOS/nixos-hardware #162: Touchpad stop working suddenly

ImExtends, 2020-05-08:
Hello, I've been using NixOS for a week, but I still have a problem: my touchpad seems to stop working from time to time for almost 3 minutes, and I have this problem like 7 times a day and all I can do is wait until he reworks again. I specified `services.xserver.libinput.enable = true;` in my `/etc/nixos/configuration.nix`, but I still have the same problem.
Did anybody encouter this issue and knows how to solve the problem ? I would be glad, thanks.

ehmry, 2020-05-08:
I've seen similar problems with a dell latitude 3480, but no fix.

ImExtends, 2020-05-08:
> I've seen similar problems with a dell latitude 3480, but no fix.

I've seen another issue with this but no fix, it would be good to have an answer, otherwise I'll have to go with another distro unfortunately 😕

ehmry, 2020-05-08:
Do you know that the touchpad works with other distros?

ImExtends, 2020-05-08:
> Do you know that the touchpad works with other distros?

Yup, I was on Arch Linux for 6-7 months and no problem with that

Moredread, 2020-05-08:
Some profiles blacklist the psmouse module (I think), is that loaded on your systems?

ImExtends, 2020-05-08:
How can I see if he is loaded?

ehmry, 2020-05-08:
`lsmod | grep psmouse`

ImExtends, 2020-05-08:
> `lsmod | grep psmouse`
> 
> 

Yeah it returns nothing

ImExtends, 2020-05-10:
No real fix, I had to reinstall it completely to make the touchpad work correctly

Moredread, 2020-05-10:
What did you reinstall?

ImExtends, 2020-05-10:
> What did you reinstall?

I didn't know how to resolve the problem, didn't find anything on internet. Plus, I had others problems including this

Mic92, 2021-07-27:
Maybe it got fixed by a kernel upgrade.

closed 2021-07-27 by Mic92