NixOS/nixos-hardware #171: Requesting configuration for Thinkpad A485

jnoortheen, 2020-06-13:
I am new to NixOS Community, finally made my jump from manjaro to nix. I am not able to find a profile ready to be used with my laptop Thinkpad A485. I am using `Thinkpad T495 ` since they beling to same lineup. I am facing the problem where once the system is put to sleep, it will not wake up. It is a request to add one config for A485.

jonringer, 2020-06-13:
If you're able to to do `Ctrl+Alt+F1` to get to the tty, this is likely a video driver issue.

jonringer, 2020-06-13:
Seeing as this is an AMD laptop, make sure your kernel is >=5.4

jnoortheen, 2020-06-13:
> If you're able to to do `Ctrl+Alt+F1` to get to the tty, this is likely a video driver issue.
It is getting restarted when trying to wake from suspend mode. 

> Seeing as this is an AMD laptop, make sure your kernel is >=5.4
Yes it is using 5.4.2 kernel version

jnoortheen, 2020-06-13:
researching on GitHub gave this link. https://github.com/savannidgerinel/nixos-thinkpad-x395

It seems to be solving the issue. I have to study it more to understand though.

Mic92, 2020-06-21:
If you can work out the pieces that fixes suspend, you can put them into this repository.

jnoortheen, 2020-06-21:
I haven't gotten to that point yet. Just made the system bootable with all the softwares I need. Still figuring out how to override the nixpkgs to get the latest ones. I will create a PR if I make it to work.

jnoortheen, 2021-06-13:
> Any updates?

Now I am in Manjaro and using nix as package manager. sleep is working in recent versions. I am not sure what is the reason.

Mic92, 2021-07-25:
Seems like there is no one with the hardware.

closed 2021-07-25 by Mic92