NixOS/nixos-hardware #173: Dell XPS 13 2-in-1 (7390) WiFi won't work with current (5.3) kernel

jpetazzo, 2020-06-21:
Hi,

(First of all, this is my first time using NixOS on a physical machine - as opposed to VM or container - so I'm sorry if this is not the right way/place to report this!)

I just installed NixOS on a Dell XPS 13 2-in-1 (apparently that's the "7390" model). I used the minimal installation CD (from [here](https://nixos.org/download.html)). The WiFi interface (a "Killer Wi-Fi 6 AX1650i 201NGW") wasn't recognized. I installed over Ethernet, and after rebooting on the newly installed system, I noticed the following message in the kernel log:

```
[    3.829133] iwlwifi 0000:00:14.3: Collecting data: trigger 15 fired.
[    4.080199] iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
[    4.080221] iwlwifi 0000:00:14.3: Status: 0x00000000, count: -1090658721
[    4.080246] iwlwifi 0000:00:14.3: Loaded firmware version: 50.3e391d3e.0
[    4.080276] iwlwifi 0000:00:14.3: 0xFC7539BB | ADVANCED_SYSASSERT          
...
[    4.082177] iwlwifi 0000:00:14.3: 0x05B0905B | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
[    4.082233] iwlwifi 0000:00:14.3: 0x0000025B | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
[    4.082352] iwlwifi 0000:00:14.3: SecBoot CPU1 Status: 0x5d55, CPU2 Status: 0x3
[    4.082375] iwlwifi 0000:00:14.3: Failed to start RT ucode: -110
[    4.082437] iwlwifi 0000:00:14.3: Firmware not running - cannot dump error
[    4.094234] iwlwifi 0000:00:14.3: Failed to run INIT ucode: -110
```

(Full kernel log in an attached file.)

After looking around, I found a few messages ([here](https://bbs.archlinux.org/viewtopic.php?id=251430), [here](https://bugzilla.redhat.com/show_bug.cgi?id=1788150), [here](https://bugzilla.kernel.org/show_bug.cgi?id=205425)) suggesting that this is a regression in 5.4 kernels, but supposedly fixed. The kernel that I was running was `5.4.46 (nixbld@localhost) (gcc version 9.2.0 (GCC)) #1-NixOS SMP Wed Jun 10 18:24:58 UTC 2020`.

Since this looks like a recent 5.4 kernel, I thought it would include the fix mentioned above; but just in case, I wanted to try to downgrade to a 5.3 kernel. I couldn't find how to do that, however I saw 5.5 and 5.6 kernels on NixOS, so I added `boot.kernelPackages = pkgs.linuxPackages_5_6;` in my `/etc/nixos/configuration.nix`, reconfigured, rebooted, and lo and behold, now the WiFi interface is recognized correctly (and after fiddling with wpa_supplicant, I got it to connect to my network).

I see [this directory](https://github.com/NixOS/nixos-hardware/tree/master/dell/xps/13-7390) with configuration for this specific laptop. Would it make sense to add something there to use the 5.6 kernel? Or, since I expect that the "main" kernel will eventually work, should it be a note in the README there instead?

(For what it's worth, when I ordered that laptop, there was no way to get another WiFi interface with it, so I _think_ it's the only WiFi interface that can be installed in it, but I'm not 100% sure either.)

Thank you!

[kern.log](https://github.com/NixOS/nixos-hardware/files/4809276/kern.log)

Mic92, 2020-06-21:
If a kernel upgrade fixes the wifi, we can add something like this to dell xps profile:

```nix
{
  boot.kernelPackages = lib.mkIf (lib.versionOlder pkgs.linux.version "5.4") pkgs.linuxPackages_latest;
}
```

I assume you are using `<nixos-hardware/dell/xps/13-7390>` right now?

jpetazzo, 2020-06-21:
Yes:

```
[jp@kater:~]$ cat /etc/nixos/configuration.nix 
{ config, pkgs, ... }:

{
  imports =
    [ 
      <nixos-hardware/dell/xps/13-7390>
      # Include the results of the hardware scan.
      ./hardware-configuration.nix
    ];
...
```

Mic92, 2020-06-22:
Ok. If you know the lower kernel version that still works, we can add it.

jpetazzo, 2020-06-22:
So:

- `boot.kernelPackages = pkgs.linuxPackages_5_6` works
- `boot.kernelPackages = pkgs.linuxPackages_5_4` doesn't work (same error message in the kernel logs)
- `boot.kernelPackages = pkgs.linuxPackages_5_5` tells me that `The 5.5 tree of Linux has been removed from 20.03.`

Would you like me to do more specific tests?

Mic92, 2020-06-23:
@jpetazzo can you test this? https://github.com/NixOS/nixos-hardware/pull/175

jpetazzo, 2020-06-24:
I edited `/etc/nixos/configuration.nix`, and I:
- removed `boot.kernelPackages = pkgs.linuxPackages_5_6;`
- added `boot.kernelPackages = lib.mkIf (lib.versionOlder pkgs.linux.version "5.6") pkgs.linuxPackages_latest; `
- added `lib` to the first line
Then after `nixos-rebuild boot` and `reboot`, I still have kernel 5.6 and WiFi works.

:tada: 

Or did you want me to test specifically with the updated `<nixos-hardware/dell/xps/13-7390>` ? (I was reading the `nix-channel` manpage and the [Nix Channels](https://nixos.wiki/wiki/Nix_channels) docs and I didn't find if there was a way to say "use this specific branch/PR for a given channel" instead of the release zip/tarball on GitHub?)

In any case, thanks a lot :)

Mic92, 2020-06-24:
Should be fair enough. Merged.

closed 2020-06-24 by Mic92