NixOS/nixos-hardware #179: Flakes: also export the common modules?

wmertens, 2020-07-25:
I have a Lenovo C940 and it's pretty similar to the X1 but still different, and I want to build my own. However, I can't use flakes for that, I can't reach the intel, ssd etc modules.

@Mic92 ?

ehmry, 2020-07-25:
On Saturday, 25 July, 2020 9:24:40 AM CEST, Wout Mertens wrote:
> I have a Lenovo C940 and it's pretty similar to the X1 but 
> still different, and I want to build my own. However, I can't 
> use flakes for that, I can't reach the intel, ssd etc modules.

When a flakes reference is coerced to a string, it becomes the
root of the flake repository, and from that you can reference
arbitrary files:

```nix
let nh = builtins.getFlake "nixos-hardware";
in [
  (nh + "/common/pc/laptop/ssd")
  (nh + "/common/cpu/intel")
]

```

wmertens, 2020-07-25:
Not quite user friendly, but good enough :) should I choose this, or perhaps the readme should be adjusted? (The url is wrong anyway)

Mic92, 2020-07-25:
@wmertens Yes we can add those. Feel free to open a PR.

marius851000, 2020-09-14:
I added the possibility to use common/* with flake in https://github.com/NixOS/nixos-hardware/pull/190

closed 2020-09-15 by Mic92

wmertens, 2021-12-01:
@ehmry is what you wrote still possible? When I do `builtins.getFlake "nixos-hardware"` it complains about impurity.

When I pass `inputs` in `extraArgs` on the flake outputs evaluation, I get an infinite recursion when trying to use `inputs`.

ehmry, 2021-12-03:
I don't do that anymore, so I'm not sure if it works, but I have been using `extraArgs` to pass flake inputs to nixos modules, within my flaked configurations.

> I'm actually surprised this evaluates, because I forgot to remove nixos-hardware from the function arguments and it still works - are flakes injected by default?

I think its possible put arbitrary named arguments on a module if they are not used.