NixOS/nixos-hardware #182: Importing <nixos-hardware/dell/xps/15-9500/nvidia> fails

alvatar, 2020-08-01:
System: Nixos 20.03

Output when importing this configuration:

```
building Nix...
building the system configuration...
error: The option `hardware.nvidia.prime' defined in `/nix/var/nix/profiles/per-user/root/channels/nixos-hardware/common/gpu/nvidia.nix' does not exist.
(use '--show-trace' to show detailed location information)
```

<nixos-hardware/dell/xps/15-9500> works well

Mic92, 2020-08-01:
That option does not exists yet in 20.03. We have it in nixos-unstable though.
cc @bahaynes

alvatar, 2020-08-02:
oh ok. It seems that hardware.nvidia.optimus_prime does exist. Is that an alternative?

I'm pretty new to NixOS. Would it be safe for me to upgrade to 20.09?

Mic92, 2020-08-02:
> oh ok. It seems that hardware.nvidia.optimus_prime does exist. Is that an alternative?
> 
> I'm pretty new to NixOS. Would it be safe for me to upgrade to 20.09?

I have not checked if it provides equivalent features. You can try to upgrade to unstable and if it not works out just roll-back to 20.03. We don't have released 20.09 yet.

alvatar, 2020-08-02:
I will give a shot to updating to the unstable channel.
Thank you!
(should I just close this or keep it for potential compatibility with 20.03?)

bahaynes, 2020-08-02:
> oh ok. It seems that hardware.nvidia.optimus_prime does exist. Is that an alternative?
> 
> I'm pretty new to NixOS. Would it be safe for me to upgrade to 20.09?

That should work, it may give deprication warnings. Before upgrading to unstable I was just using the NVIDIA sync (there is an example in the 9560 folder), but it has worse battery performance.

alvatar, 2020-08-02:
Thanks @bahaynes

Mic92, 2020-08-03:
Should we keep this open?

bahaynes, 2020-08-04:
> Should we keep this open?

I'm not sure how we'd handle alternate configuration for 20.03 (is there a way to detect that?). It also will be deprecated when 20.09 is released. I could just update the readme to point to the 9560 profile, which is set up to use sync.

Mic92, 2020-08-04:
> > Should we keep this open?
> 
> I'm not sure how we'd handle alternate configuration for 20.03 (is there a way to detect that?). It also will be deprecated when 20.09 is released. I could just update the readme to point to the 9560 profile, which is set up to use sync.

Right now have some version checks for some option, but testing on stable is not very good. I hope I can improve the situation here a bit more: https://github.com/NixOS/rfcs/pull/70

Mic92, 2021-06-13:
Should be no longer an issue.

closed 2021-06-13 by Mic92