NixOS/nixos-hardware #193: Dell XPS 13 7390 Settings - Frequently freezing screen

tfc, 2020-10-02:
Hi there,

cc @Alecktos @Mic92 as you seem to have this laptop model, too (deducing that from your collaboration on this file https://github.com/NixOS/nixos-hardware/blob/master/dell/xps/13-7390/default.nix)

I have frequent screen freezes in pretty random situations/frequency. (Yesterday with a specific USB c hub it was a freezing screen roughly every minute for 1 second).
After going through some forums and finding experiences/fixes of other users, i got this fixed by setting:

```nix
            boot.kernelParams = [
              "intel_idle.max_cstate=1"
              "i915.enable_dc=0"
            ];
```

With this setting the freezing is gone and the whole laptop feels much more responsive.

Is that something that other 7390 users experienced (on linux / NixOS), too, or is my experience here completely odd?
If it's not just me i'd open a pull request on that file adding this setting.

Alecktos, 2020-10-05:
Hi!
I use the same laptop model but have not experience any freezes (NixOs, i3 setup).

Mic92, 2020-10-16:
@tfc Maybe add it as tip to the `README.md` of this profile and reference this issue until we can confirm this is a more common issue.

jakeisnt, 2021-01-02:
I use the same laptop and haven't had any issues with my current configuration (NixOS, Wayland/Sway). Do you have any information regarding the specific USB-C hub you've use? I've definitely had display issues with USB-C hubs in the past, but I wasn't using NixOS at the time and don't have access to those hubs anymore. 

It's definitely worth mentioning the issue in the README but I fear that these settings (like the enable_dc=0) may hurt more than help if they're added generally.

tfc, 2021-01-06:
Hey, thank you for your input!

The laptop started with more random crashes etc. and after a lot of googling my problems i got the impression that there are many very buggy specimen of the DELL XPS 13 7390 laptop model. So i lost my hope with my specimen and stopped using it.

I close this issue as @Alecktos and @jakeisnt seem to be happy with the current module implementation and it would not make sense to consider buggy specimen of this laptop model.

closed 2021-01-06 by tfc