NixOS/nixos-hardware #211: apple/macbook-pro/12-1: won't resume after hibernation

sevenfourk, 2020-11-25:
not much details now, todo.

Mic92, 2020-11-26:
Just to clarify. The swap is big enough. Is there disk encryption used?

sevenfourk, 2020-11-26:
added some description, but you're right, that's all upstream issues, same as my another issue here, concerning wireless.

apparently there are lots of issues with specifically `brcmfmac` driver (or the hardware itself).

pretty old bug, but i guess still relevant:

https://bugzilla.kernel.org/show_bug.cgi?id=101681#c116

because i had success resuming after hibernation doing: `rmmod brcmfmac && systemctl hibernate`.

Mic92, 2020-11-26:
We could try to automate this with systemd suspend modules: https://gist.github.com/mauron85/f764c88d2dc47af157487a7a318dd35b

PR #214 (2020-12-01): apple/macbook-pro/12-1: PM: set cpuFreqGovernor and power Up/Down commands
fixes #211.
fixes #210.

Added default CPU governor "conservative" to enable gradually increasing/decreasing CPU freq, rather than using "powersave", which would keep CPU freq at 0.8Ghz.

 Added power Up/Down PM commands which will unload and load brcmfmac on state change. This is not elegant way, to say the least.
closed 2020-12-01 by Mic92