NixOS/nixos-hardware #219: NVIDIA Jetson Nano and Xavier NX support

danmx, 2020-12-27:
I'm still starting with NixOS but I find it very useful to have a declarative system configuration. I saw that it's possible to build custom NixOS images using [nixos-build and hydra](https://nixos.mayflower.consulting/blog/2018/09/11/custom-images/).

Since NVIDIA provides its [JetPack SDK](https://developer.nvidia.com/embedded/jetpack) Ubuntu-based image. Would it be possible to convert it to NixOS?

Mic92, 2020-12-28:
Maybe @samueldr can help out here on how to port NixOS to this device.

blakat360, 2021-06-03:
I would also be interested in knowing how to do this.

samueldr, 2021-06-03:
I have no information about NVIDIA hardware other than "third party" data.

AFAIUI they use a custom boot chain, which makes things harder to port.

And AFAICT there might be enough relevant differences in the different NVIDIA platforms that solving it for one wouldn't be enough for the others. Though take this with a grain of salt, I'm working from a flawed **superficial** understanding.

Sadly without the hardware to look at, prod and poke, there's likely nothing I can really do.

blakat360, 2021-06-05:
@samueldr I'd be happy to send you over a nano to play with if that would help?

samueldr, 2021-06-05:
Only if you were able to send *time* along with it. Sadly it would likely sit unchecked like some other things I acquired for my own use...

I don't really have the free time to look at this *on top* of everything else I'm doing, sorry :/...

danmx, 2021-06-05:
Is it possible to get NVIDIA's kernel and bootloader with just NixOS' userspace?

Mic92, 2021-06-13:
@danmx Yes. You can build nixos images without kernel/initrd. Here an example:

This builds an ext4 image: https://github.com/Mic92/vmsh/blob/main/nix/nixos-image.nix 
And here is the corresponding nixos configuration that builds an init executable that you can boot from:
https://github.com/Mic92/vmsh/blob/d8ac04fc856c08a98021d1afd9287542eeb2eeb9/nix/modules/configuration.nix#L15

This is an example how you can boot it with qemu: https://github.com/Mic92/vmsh/blob/d8ac04fc856c08a98021d1afd9287542eeb2eeb9/justfile#L137

blakat360, 2021-12-09:
@Mic92 please do. How do we proceed?

Mic92, 2021-12-10:
> @samueldr I'd be happy to send you over a nano to play with if that would help?

If this over is still valid and it arrives before I go on Christmas holiday, I would try to get it running. I would probably go for the low hanging fruit by re-using their kernel/boot loader and only provide the nixos userland. In case you are still interested you can reach me at @mic92:nixos.dev on matrix.

blakat360, 2021-12-10:
do you have discord?

Mic92, 2021-12-10:
Yes. Mic92#0022 in the nixos discord channel.

kjeremy, 2022-02-16:
I'm curious if anyone had a chance to play with this.

Mic92, 2022-02-17:
Nope. Still don't have it.

RyzeNGrind, 2022-11-16:
I have a NVIDIA Jetson AGX Xavier 32GB and would be interested in contributing towards the development and deployment of NixOS on the NVIDIA Jetson SDK Based Ubuntu image, if there are others interested perhaps we could collaborate on our NVIDIA findings? I currently have a cluster of arm64 RPI4s that will be running NixOS soon as well.

danielfullmer, 2022-12-01:
Hey all, I've been able to open-source (jetpack-nixos)[https://github.com/anduril/jetpack-nixos], which is a module that enables NixOS to run on Nvidia Jetson devices.

Announcement post: https://discourse.nixos.org/t/jetpack-nixos-nixos-module-for-nvidia-jetson-devices/23632

This includes the Xavier AGX, Xavier NX, and Orin AGX, but unfortunately not the Jetson Nano, since support was dropped by Nvidia in the latest Jetpack release.

Mic92, 2023-03-01:
I think this project is than good enough. We could potentially integrate this into nixos-hardware but if the author prefers to keep it seperate that's also fine.

closed 2023-03-01 by Mic92

danielfullmer, 2023-03-05:
I'm certainly open to merging this into nixos-hardware (or potentially nixpkgs itself--to unify CUDA packaging).  But, I think we should wait and look again when Jetpack 6 is released (planned to be Q3 2023):

From https://forums.developer.nvidia.com/t/jetson-software-roadmap-for-2023/238889
> Outlined in last year’s roadmap announcement, NVIDIA will finish upstreaming Jetson support to the Linux Kernel by JetPack 6 and pave the way for greater choice of Linux based distributions on Jetson.

ShalokShalom, 2023-07-03:
> I'm certainly open to merging this into nixos-hardware (and parts of it to nixpkgs itself--to unify CUDA packaging). But, I think we should wait and look again when Jetpack 6 is released (planned to be Q3 2023):
> 
> From https://forums.developer.nvidia.com/t/jetson-software-roadmap-for-2023/238889
> 
> > Outlined in last year’s roadmap announcement, NVIDIA will finish upstreaming Jetson support to the Linux Kernel by JetPack 6 and pave the way for greater choice of Linux based distributions on Jetson.

Sorry, you can clear me up: How is waiting for JetPack 6 helping, when 5 already dropped support?

danielfullmer, 2023-07-08:
What are you referring to specifically? The Jetson Nano? TX2? If so, you're right, waiting for Jetpack 6 would not help those devices because NVIDIA dropped support in Jetpack 5.  Jetpack 6 would only affect Orin devices.

I should mention, however, that there is some level of support for Tegra devices in the mainline Linux kernel. I know someone who has booted NixOS w/ a mainline kernel on a TX2 devkit and has tested pcie and ethernet as working there.  Obviously CUDA would be unlikely to work, but NVDEC for Tegra is in the mainline kernel: https://www.phoronix.com/news/NVIDIA-Tegra-Linux-5.16-NVDEC

mariobaldini, 2024-03-16:
Hi Folks, 

I'm also interested on this (specially for jetson nano). 
Anyone still working on it so I could try to collaborate? 

Thanks