NixOS/nixos-hardware #234: XPS 13 (9310) graphics issues

rowanG077, 2021-03-02:
I have been following the [pull request](https://github.com/NixOS/nixos-hardware/pull/207) and now audio, bluetooth and wifi work! Which is very great :)!

but I have graphics issues which make the laptop essentially unuseable. I have a model with the  i7-1185g7 CPU. I use i3+xfce and I'm on the current nixos-unstable branch.

On first boot I get to the login screen fine but after logging in the problems start. With the exception of the mouse cursor the screen is frozen. I do see the background and the i3 status bar but it does not update. When I switch to a terminal and back to the graphical interface the screen does start updating. 

But it's only updating when the layout changes in i3, For example when I open my main terminal (alacritty) it does open the window but shows a blank terminal. When I open another app or terminal. The old terminal updates once but is then frozen again. Other apps behave similarly. See [this attached video](https://user-images.githubusercontent.com/7439756/109692597-02a47380-7b89-11eb-82ae-cad3275c4398.mp4) for an example of this.

This is the behaviour without a compositor. If I enable picom. The apps do start to get updated with changing i3 but the updates are slow and partial. See [this attached video](https://user-images.githubusercontent.com/7439756/109692689-1c45bb00-7b89-11eb-9c9b-27ad25ec877f.mp4) for an example where `glxgears` is running.

I have no real clue why this is happening but it seems to be some kind of optimization that is happening. Looking online I couldn't find anyone with the same issues as me but there where some people that had issues with garbage data on the screen and I tried some of the fixes for that without succes.

What I tried so far:

- `i915.modeset=1`, `i915.fastboot=1` and `i915.enable_psr=1` kernel boot params together and every single one by itself. I had hoped `i915.enable_psr` would do the trick since that is an optimiziation that stops the laptop from refreshing the screen itself.
- Enable disable picom.
- Setting `hardware.cpu.intel.updateMicrocode` to `false`.

I really have no more shots left at this point and hope someone here has any clue what could be going on.

Note that (prior to wiping) my laptop worked fine in windows so I'm quite sure it's not a hardware fault.

rowanG077, 2021-03-02:
Some more things I have tried with no success:

- `Option      "AccelMethod"  "uxa"` in the xserver config. This won't load any UI at all with dmesg errors about pipe underruns.
- adding `i915.enable_dc=0` or `intel_idle.max_cstate=4` as suggested [here](https://bugs.freedesktop.org/show_bug.cgi?id=103229)

rowanG077, 2021-03-03:
Sorry to ping you @mitchmindtree ^^. But would you have any idea?

mitchmindtree, 2021-03-03:
Hi @rowanG077! I've not run into any of these issues myself so unfortunately I'm not sure how much I can help. I'm running a pretty stock GNOME experience with multiple displays at different DPIs and it's running surprisingly fawlessly at the moment. I had one graphics issue a while ago (much earlier in the boot sequence and more severe) which turned out to be a bug in the Intel microcode, but that has long since been updated.

I wonder if perhaps the bug is something to do with that particular version of the kernel and your desktop environment? Maybe it's worth trying to switch to (or from) the `nixos-unstable` channel?

Btw sidenote: I also have this in my personal config:

```nix
  # Enable Intel's Hybrid Driver.
  nixpkgs.config.packageOverrides = pkgs: {
    vaapiIntel = pkgs.vaapiIntel.override {
      enableHybridCodec = true;
    };
  };
```

The only other option that comes to mind is maybe updating firmware using `fwupdmgr`? Here's my output of `fwupdmgr get-devices` which hows the current versions on my system:

```
XPS 13 9310
│
├─USB3.0 Hub:
│     Device ID:          cf294bf55b333004beb7c41f952c1838c23e1f4a
│     Summary:            USB 3.x Hub
│     Current version:    3.83
│     Vendor:             VIA Labs, Inc. (USB:0x2109)
│     Install Duration:   15 seconds
│     GUIDs:              95d09348-73f7-5964-ae90-54c6b1ad5bbf
│                         26f33695-3a3e-5c08-badb-f6141390ebd9
│                         10eb3a15-c177-5810-af53-1963e9200e65
│                         97840b50-84f3-5d44-9d5b-524f5ff06af8
│                         af708d09-54b5-5f8d-8c6a-a0fe0f91cce5
│                         022d2f73-4826-546a-ba0f-62579ea848ea
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│   
├─KXG60PNV2T04 NVMe KIOXIA 2048GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express Solid State Drive
│     Current version:    10704104
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     GUIDs:              73a0c81f-2555-56a5-bb6a-c2aeab2213d1
│                         558810bc-de64-11e8-9f32-f2801f1b9fd1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          070a7ea3ab21dbc8d5c913fe39e3fb636b985492
│ │   Current version:    1.1.4
│ │   Minimum Version:    1.1.4
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              cfb08d7c-8d3d-41b1-846b-48f634dfac9d
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       967c55a7-3ef5-517d-85be-61f95651696a
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
[... 46 lines cut ...]
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─USB2.0 Hub:
      Device ID:          1bcca71b3480ec5662e5da8a742c4c805275ffbf
      Summary:            USB 2.x Hub
      Current version:    3.83
      Vendor:             VIA Labs, Inc. (USB:0x2109)
      Install Duration:   15 seconds
      GUIDs:              db31bd61-b70f-553d-a94f-c579fa11bf40
                          3f0581b6-15ed-5f87-91eb-296db71e0838
                          d390a7f5-ea48-5370-87a7-eba89bf5606e
                          62cc1594-f280-568e-9e50-3739cbd40cca
                          d42e8a1e-2315-5f19-8f1d-2ad948355084
                          8453111a-0e73-51f4-877f-fa340d8b2c03
      Device Flags:       • Updatable
                          • Cryptographic hash verification is available
                          • Device stages updates
                          • Device can recover flash failures
```

---

I didn't have time to A/B it and test it to see if it still has any effect, but I don't think it is enabled by default and whether or not to add it to the `xps-9310` config has been on the back of my mind. That said, I'm not sure it's even related to your issue, I think it might just be to do with video decoding acceleration.

rowanG077, 2021-03-03:
Thank you very much! I didn't think desktop environment would impact this.

The live cd had this issue as well so I will make a few different images to see of gnome and plasma have this issue as well.

rowanG077, 2021-03-03:
Gnome freezes after login, possible the same issue as i3 but hard to say since switching to a terminal and then back to the UI shows the login screen again.

Plasma boots without issues and auto logins. I don't have any graphical issues here. That's good to know that it's not a fundamental issues. I will dig deeper into the configuration differences between plasma and gnome.

rowanG077, 2021-03-03:
I checked DRI 2 and DRI 3 both have the same issue.

rowanG077, 2021-03-03:
The issue is gone when I force xserver to use the `modesetting` driver.

```
services.xserver.videoDrivers = [ "modesetting" ];
```

I see that the intel drive is quite old in even unstable channel. I will try to update it and see if that helps.

rowanG077, 2021-03-03:
@mitchmindtree what x video driver do you use? Should this maybe be added to the `xps-9310` config?

mitchmindtree, 2021-03-03:
Hmmm not sure, this is the only `xserver` stuff in my config:

```nix
  services.xserver.enable = true;
  services.xserver.displayManager.gdm.enable = true;
  services.xserver.desktopManager.gnome3.enable = true;
```

That said, I've run the default GNOME which uses Wayland for the last year - does that still use the xserver? I think it uses a different stack entirely maybe? I don't quite remember.

rowanG077, 2021-03-03:
Do you run unstable @mitchmindtree?  It's possible that the issue I'm having only shows itself on unstable if so.

mitchmindtree, 2021-03-03:
I'm also on unstable:

```
[mindtree@mindtree:~]$ sudo nix-channel --list
home-manager https://github.com/nix-community/home-manager/archive/release-20.09.tar.gz
nixos https://nixos.org/channels/nixos-unstable
nixos-hardware https://github.com/NixOS/nixos-hardware/archive/master.tar.gz
```

rowanG077, 2021-03-03:
That's odd I tried the gnome live cd environment and I couldn't even boot into it. That's with the patched kernel. Do you have the same cpu as me(i7-1185g7)?

mitchmindtree, 2021-03-03:
[image]

Looks like I have the 1165G7 - not quite sure what the difference is?

rowanG077, 2021-03-03:
Not much different. Slightly higher clock speed for CPU and GPU. But nothing where I would say things should be incompatible... I guess let's close this issue for now. There were multiple people for whom the current changes work.

closed 2021-03-05 by rowanG077