NixOS/nixos-hardware #238: Build Custom Hardware Kernel on build Server

busti, 2021-03-14:
I am trying to build a custom kernel for my Microsoft Surface device on a build server (which is also a binary cache) to improve turnaround times in setting up my laptop.

I have added the hardware channel as an unprivileged user on my build server and I also ran nix-channel --update as the same user.

I have tried to nix-build '<nixos-hardware/microsoft/surface>' but it errors with:

error: cannot auto-call a function that has an argument without a default value ('config')
People in the telegram group are telling me I should clone the hardware repo and remove the config parameter, but that seems excessive to me. When used in an os config environment "${builtins.fetchGit { url = "https://github.com/NixOS/nixos-hardware.git"; }}/microsoft/surface" it also builds somehow.
How can I make this build manually?

closed 2021-03-14 by busti

devins2518, 2021-08-29:
@busti hey, were you able to find out how to build this? I'm facing the same issue, building the kernel on 4 threads is a bit painful :sweat_smile:

Mic92, 2021-08-30:
The ci script of this repo almost does this for you, you just need to remove `--dry-run` from here: the https://github.com/NixOS/nixos-hardware/blob/342048461da7fc743e588ee744080c045613a226/tests/run.py#L55

In my dotfiles I also build all my machines in CI: https://github.com/Mic92/dotfiles/blob/master/outputs.nix#L96
I currently don't have a good example for a pre-flake CI.

busti, 2021-08-30:
@devins2518 Yeah, I have two methods of building my system on my build server and use whichever works the best right now.  
For one I have added a custom build machine in my systems nixos config:  
```
    buildMachines = [{
      hostName = "builder";
      system = "x86_64-linux";
      maxJobs = 1;
      speedFactor = 2;
      supportedFeatures = [ "nixos-test" "benchmark" "big-parallel" "kvm" ];
      mandatoryFeatures = [ ];
    }];
```  
With this config whenever I run `nix-build` or `nixos-rebuild`, etc. it will check if the machine with the hostname builder in my local network is running and use it to build derivations instead.  

The second method I use is to build my complete system closure on the build machine using the following build command:  
```
$ nix-build --attr system "./nixos.nix" -o result-closure
```
where `./nixos.nix` is
```
let
  nixpkgs = builtins.fetchTarball {
    url = "https://github.com/NixOS/nixpkgs/archive/b74b1cdb2fecc31ff7a127c5bc89771f887c93bb.tar.gz";
    sha256 = "0ncr4g29220amqm4riaa1xf4jz55v2nmh9fi16f1gzhww1gplk8h";
  };

in
  import "${nixpkgs}/nixos" {
    configuration = {
      imports = [
        ./configuration.nix
      ];
    };

    system = "x86_64-linux";
  }
```  

Mostly taken from this discourse thread:  
https://discourse.nixos.org/t/is-it-possible-to-have-nix-install-base-system-and-desktop-reboot-and-then-have-it-install-everything-else-in-the-background/4408?u=busti