NixOS/nixos-hardware #240: x1 nano hardware support issues

klautcomputing, 2021-03-19:
# Hardware
x1 nano 20UN0009US
## WiFi
works when updating to a newer kernel, I am on 5.11.6 now
## Inputs
### trackpoint + wheel
got the the wheel scrolling to work with:
```
    # x1 trackpoint
    trackpoint = {
      enable = true;
      device = "TPPS/2 Elan TrackPoint";
      # middle mouse button scroll
      emulateWheel = true;
    };
```
Setting the `device` is what made the difference for me.
### Fn Keys
The only one that works for me is the :no_entry_sign: :signal_strength: one, useful for when you don't want to have wifi...
So neither sound, nor brightness controls work out of the box
### Touchpad
No idea, I don't use it. disabled it in bios before the first boot.
## Screen
### Camera
works
### Brightness
I cannot control screen brightness, neither with `light` nor with the Fn keys (they don't do anything)
## Bluetooth
works
## Fingerprint reader
:shrug: 
## Audio
Speaker, headphone jack + bluetooth work out of the box

## NixOS info
FYI:
```
 - system: `"x86_64-linux"`
 - host os: `Linux 5.11.6, NixOS, 21.05pre276515.1f77a4c8c74 (Okapi)`
 - multi-user?: `yes`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.3.10`
 - channels(root): `"nixos-21.05pre276515.1f77a4c8c74"`
 - channels(leex): `"home-manager-20.09, nixpkgs-unstable-21.05pre276390.2996d6df1de"`
 - nixpkgs: `/nix/var/nix/profiles/per-user/root/channels/nixos`
 ```

Should I create a new config file and contribute the scrolling and the minimum kernel version to the project?

Mic92, 2021-03-19:
Sure go ahead. Regarding the kernel version you could set it optionally if the default kernel is too old. Something like: `boot.kernelPackages = lib.mkIf (lib.versionOlder pkgs.linux.version "5.6") pkgs.linuxPackages_latest;`

klautcomputing, 2021-03-20:
FYI https://old.reddit.com/r/thinkpad/comments/m86650/is_anyone_running_a_linux_distros_on_the_x1_nano/ other distributions don't have issues with the Fn keys. Any idea what's going on here?

klautcomputing, 2021-03-20:
Update: `brightnessctl` works fine!

weihua-lu, 2021-03-25:
@klautcomputing  
Same model here, I'm on the linuxPackages_latest(5.11.7) with gnome, everything seems to work fine by default, i don't have issues with fn keys.
However, when using headphone, there's some constant noticeable hissing background noise, tried to tweaked alsamixer with no luck, are you experiencing the same maybe?

weihua-lu, 2021-03-25:
Update: 
It's not a nano specific issue it seems. This magic command from arch wiki would fix the headphone issue:
`sudo hda-verb /dev/snd/hwC0D0 0x1d SET_PIN_WIDGET_CONTROL 0x0`
https://wiki.archlinux.org/index.php/Lenovo_ThinkPad_X1_Carbon_(Gen_7)

nashamri, 2021-03-27:
@klautcomputing Would love to see x1 nano supported :+1:

nashamri, 2021-03-27:
I also had an issue with sound not working at all, and this fixed it for me:
`  boot.kernelParams = [ "snd_hda_intel.dmic_detect=0" ];`

shuhaowu, 2021-04-30:
@klautcomputing :wave: :stuck_out_tongue: are you seeing issues where your computer freezes sometimes and it seems to correlate during shutdown? I'm not quite using the same setup as you but am curious...

klautcomputing, 2021-04-30:
@shuhaowu :wave: Long time no see!

I had that exactly once, and only when I triggered the reboot without sudo from my user account and an older kernel as well. I will report back if it pops up again.

shuhaowu, 2021-04-30:
Long time no see indeed!

I've had the problem a few times, and at least once it happened immediately after I pressed "shutdown" in gnome. I had to do a hard reset and nothing interesting was left behind in the logs. I'm on 5.8. I may try updating to 5.11 or 5.12 at some point and see if it happens again.

supermarin, 2022-12-29:
Another x1 nano long term user here. Everything works without using nixos-hardware on my end, a couple caveats:
- Bluetooth kb/mouse is suffering the autosuspend problem (stops reacting when you leave it idle for a few sec. Comes back to life after 1-2 seconds of lag)
- dpms OFF works, but dpms ON doesn't bring back up the external display connected via thunderbolt (lg ultrafine 5k in my case)

Kernel 6.1, firmware up to date, nixpkgs master. Wayland + sway. Anyone else observing any of the two?

gunar, 2023-02-01:
I'm on a Gen 2

```
$ #  boot.kernelPackages = pkgs.linuxPackages_latest;
$ uname -a
Linux nixos 6.1.8 #1-NixOS SMP PREEMPT_DYNAMIC Tue Jan 24 06:24:44 UTC 2023 x86_64 GNU/Linux
```

supermarin, 2023-02-01:
Are you using nixos-hardware? What boot.kernelParams?
For the record I'm not using nixos-hardware. These are all my hw settings: https://github.com/supermarin/dotfiles/blob/main/nixos/hardware-x1.nix

gunar, 2023-02-01:
I'm not on `nixos-hardware` and also not using any `boot.kernelParams`.

gunar, 2023-02-01:
For anyone else looking to get the camera working on a Gen 2, [this wiki](https://wiki.archlinux.org/title/Lenovo_ThinkPad_X1_Nano_(Gen_2)#Camera) provides some insight. As of today, it seems there are only hacky solutions for now.

gunar, 2023-02-01:
Also this: https://github.com/NixOS/nixpkgs/pull/212799

mweinelt, 2023-02-14:
> Also this: [NixOS/nixpkgs#212799](https://github.com/NixOS/nixpkgs/pull/212799)

If anyone is aware how to wire the packages in that PR up correctly, I'd be interested to hear it.

betaboon, 2023-04-07:
for cross-linking: https://github.com/NixOS/nixpkgs/pull/225160