NixOS/nixos-hardware #243: XPS 13 9310 question

seqizz, 2021-03-27:
Hi, I saw the [related config](https://github.com/NixOS/nixos-hardware/blob/master/dell/xps/13-9310/default.nix) but I didn't need any of those when I use `boot.kernelPackages = pkgs.linuxPackages_latest;`. So my question is, is this the same hardware? If not I'd also like to add config for this model, maybe suffixing with "2-in-1". My result is:

```
~> sudo dmidecode| grep -i xps
        Product Name: XPS 13 9310 2-in-1
```

Maybe @mitchmindtree ?

Mic92, 2021-03-28:
Is it possible that these patches became upstream in one way or the other?
Maybe we can drop the whole custom kernel thing.

Mic92, 2021-03-28:
Maybe you and @mitchmindtree could both post the wifi chipset that was used.
Most patches seems to be related to this.

seqizz, 2021-03-28:
Aaah makes sense. Looks like there are [2 different variants of wifi](https://wiki.archlinux.org/index.php/Dell_XPS_13_(9310)#Wifi) on this model. Mine is indeed AX201:

```
00:14.3 Network controller: Intel Corporation Wi-Fi 6 AX201 (rev 20)
```

Since there is no "if" we can use, should I send my version with -ax201 suffix?

mitchmindtree, 2021-04-04:
@seqizz we might be better off creating a module option (see [here](https://nixos.wiki/wiki/Module) for info about module structure) for specifying either one in the configuration.

seqizz, 2021-04-07:
Honestly I don't have much experience with modules, will check it out. Hopefully there is an example in the repo already.

Mic92, 2021-04-07:
I would add an option to enable AX500 support and deprecate it once mainline supports this wifi chipset.

This is an example for a module option:

https://github.com/NixOS/nixpkgs/blob/d9f9a4a9760ef3cf3df63771cebefacd1b0178b4/nixos/modules/hardware/keyboard/zsa.nix#L23