NixOS/nixos-hardware #248: Microsoft surface module headless build errors

deliciouslytyped, 2021-04-18:
@mexisme 
If I try to import the microsoft surface module, my build fails with errors related to cairo-xlib.
I have to post the build log later, but what even depends on x11 here? How do you build your system successfully?
I think I didn't manage to even with gdm/gnome enabled.

deliciouslytyped, 2021-04-19:
https://github.com/NixOS/nixpkgs/issues/119841

mexisme, 2021-04-19:
> @mexisme
> If I try to import the microsoft surface module, my build fails with errors related to cairo-xlib.
> I have to post the build log later, but what even depends on x11 here? How do you build your system successfully?
> I think I didn't manage to even with gdm/gnome enabled.

Could it be related to something other than the Surface patches?
As I don't believe the current NIxOS+Surface patches touch anything except Kernel and Drivers;  there are some patches available for things like the X11/Wayland Wacom behaviour, but I haven't had time to port those to Nix, yet.

FYI:
I ran a `nixos-rebuild switch --upgrade` on my Surface Go over the weekend, and that seemed to work for me.
I no-longer have access to a Surface Laptop, so can't validate that side any more.

I generally just import the nixos-hardware patches for Surface:

```
# (etc)
  imports =
    [
    # Add the NixOS-hardware channel, first:
    #   nix-channel --add https://github.com/mexisme/nixos-hardware/archive/master.tar.gz nixos-hard
ware
      <nixos-hardware/microsoft/surface>
      <nixos-hardware/microsoft/surface/firmware>
      <nixpkgs/nixos/modules/installer/scan/not-detected.nix>
    ];

  boot = {
    # This is provided by the nixos-hardware channel:
    #kernelPackages = pkgs.linuxPackages_latest;
    initrd.availableKernelModules = [ "xhci_pci" "ehci_pci" "ahci" "usb_storage" "usbhid" "sd_mod" "vfat" "nvme" "rtsx_pci_sdmmc" ];
    kernelModules = [ "kvm-intel" "hid-microsoft" "hid-apple" ];
    kernelParams = [ ];
    extraModulePackages = [ ];

# (etc)
```

Everything else in my config is not Surface-specific.

deliciouslytyped, 2021-04-19:
It's not going to be a direct patches issue I just figured you might have some idea what's going on.

Mic92, 2021-04-20:
This does not look related...

closed 2021-04-20 by Mic92

deliciouslytyped, 2021-04-25:
Solved in the linked nixpkgs issue.