NixOS/nixos-hardware #260: recent nixos master fails to build due to tlp setting

wmertens, 2021-05-12:
See https://github.com/NixOS/nixpkgs/pull/97972#issuecomment-834774554

Not sure if this should be solved here or in nixpkgs.

Mic92, 2021-05-12:
Maybe would enable `config.services.tlp.enable` only if `services.power-profiles-daemon.enable ` is not enabled?

Mic92, 2021-05-12:
Right now this is only an issue if the user has enabled power-profiles-daemon them-self, right?

terlar, 2021-05-12:
That is partly true, or if they are using something that enables it, for example GNOME. Found in the linked issue:
https://github.com/NixOS/nixpkgs/commit/316928e8c15994053fb72252418d2b3197cb5dc1

I am one of the latter and not sure I would have figured it out without these two issues.

benley, 2021-08-04:
Pretty sure this is fixed by #263

closed 2021-08-04 by benley