NixOS/nixos-hardware #262: Support for additional "dtparams" for Raspberry Pi

samueldr, 2021-05-13:
It might be required at some point in the future.

Preliminary exploration.

* * *

Picking an arbitrary one from [the wiki entry](https://nixos.wiki/wiki/NixOS_on_ARM/Raspberry_Pi_4).

> When running from USB device without SD card present, kernel spams log about missing SD card, workaround for this is to set: 
>
> ```
> {
>   boot.loader.raspberryPi.firmwareConfig = "dtparam=sd_poll_once=on";
> }
> ```

It is not an overlay. So it won't work exactly like #261 outright.

Here's the description:

 - https://github.com/raspberrypi/linux/blob/73cee4f50858d1cb3834fa11d2bc7a1b4b6fe616/arch/arm/boot/dts/overlays/README#L223-L228

And here's the "definition":

 - https://github.com/raspberrypi/linux/blob/7fb9d006d3ff3baf2e205e0c85c4e4fd0a64fcd0/arch/arm/boot/dts/bcm2711-rpi-4-b.dts#L627

So, from that, and my previous reading about how `dtparams` works, I assume something similar to the following would work:

```
            /dts-v1/;
            /plugin/;
            / {
              compatible = "brcm,bcm2711";
              fragment@0 {
                target = <&emmc2>;
                __overlay__ {
                  non-removable;
                };
              };
            };
```

This is because the following override:

```
		sd_poll_once = <&emmc2>, "non-removable?";
```

Says: the `sd_poll_once` param adds the `"non-removable"` node when "true".

> Note that a property is assigned the value false by not defining it. Boolean parameters are declared like this:
>
> ```
> name = <&label>,"property?";
> ```
 - https://www.raspberrypi.org/documentation/configuration/device-tree.md

> Boolean parameters can cause properties to be created or deleted, but they can't delete a property that already exists in the base DTB.

Not that it matters for our use cases.

* * *

### References

 - https://www.raspberrypi.org/documentation/configuration/device-tree.md

samueldr, 2021-05-13:
AFAIK this would be needed for `audio` too.

 - https://github.com/raspberrypi/linux/blob/108ac8cefd05ef00bba93852e7e7f12d009674a5/arch/arm/boot/dts/bcm270x-rpi.dtsi#L92

nixos-discourse, 2021-05-17:
This issue has been mentioned on **NixOS Discourse**. There might be relevant details there:

https://discourse.nixos.org/t/boot-config-txt-related-questions-for-raspbarry-pi/3650/4

Mic92, 2021-06-13:
Has this been solved? I saw several parameters being added.

justinkb, 2023-05-09:
I have implemented this two days ago, it was pretty simple. It involves changes to three files, nixos/modules/hardware/device-tree.nix and pkgs/os-specific/linux/device-tree/default.nix in nixpkgs, and apply-overlays-dtmerge.nix from nixos-hardware/raspberry-pi/4 (ideally this apply-overlays-dtmerge.nix should be mainlined into nixpkgs).

see https://bpa.st/raw/CJUPI for the nixpkgs changes
https://bpa.st/raw/QHONW for the nixos-hardware changes

Note that the second patch also includes a little fix for allowing the overlay merging to always apply if overlays are advertised as compatible with "brcm,bcm2835", since according to the Raspberry Pi official documentation that indicates it is compatible with all products.

These changes obsolete most of the nixos-hardware/raspberry-pi/4 repo essentially, since all those manually copy and pasted overlay files (with changes to "compatible") are no longer needed. In my nixos config, I only need to import the edited apply-overlays-dtmerge.nix and then I can just use this

```
hardware.deviceTree = {
    params = [ "i2c_arm=on" ];
    overlays = [
      {
        name = "disable-bt";
        dtboFile = "${config.boot.kernelPackages.kernel}/dtbs/overlays/disable-bt.dtbo";
      }
      {
        name = "disable-wifi";
        dtboFile = "${config.boot.kernelPackages.kernel}/dtbs/overlays/disable-wifi.dtbo";
      }
      {
        name = "vc4-fkms-v3d-pi4";
        dtboFile = "${config.boot.kernelPackages.kernel}/dtbs/overlays/vc4-fkms-v3d-pi4.dtbo";
      }
      {
        name = "i2c-rtc";
        dtboFile = "${config.boot.kernelPackages.kernel}/dtbs/overlays/i2c-rtc.dtbo";
        params = [ "ds3231" "addr=0x68" ];
      }
    ];
    filter = "bcm2711-rpi-4-b.dtb";
  };
```

It'll just use the compiled dtbo overlay files that are created when building the kernel (I have also updated to the kernel to 6.1.27 in my local nixpkgs repo)

WizardUli, 2023-05-09:
I have an implementation (as a nixosModule) without requiring any nixpkgs changes but the params are defined in a separate list of attrsets and matched using name e.g.:
```
overlaysParams = [
    {
      name = "i2c-rtc";
      params = [ "ds3231" "addr=0x68" ];
    }
];
```
I could clean it up and publish it however it is currently missing dtbo compatibility checks because current implementation in this repo is IMHO plainly wrong and I haven't had time to devise a correct one.

justinkb, 2023-05-09:
Out of interest, how do you apply device tree params to the dtb base with
your nixosModule?

WizardUli, 2023-05-09:
@justinkb 
> Out of interest, how do you apply device tree params to the dtb base with your nixosModule?

Short answer: I create a different overlay overlying the `deviceTree.applyOverlays` for each different `paramsPerOverlayMap` which is created from a new option `deviceTree.overlaysParams`.

Long answer:
```nix
paramsPerOverlayMap:
{ lib, stdenvNoCC, dtc, libraspberrypi }:

with lib; (base: overlays': break (stdenvNoCC.mkDerivation {
  name = "device-tree-overlays";
  nativeBuildInputs = [ dtc libraspberrypi ];
  buildCommand =
    let
      overlays = toList overlays';
    in
    ''
      mkdir -p $out
      cd "${base}"
      true
      find . -type f -name '*.dtb' -print0 |
        xargs -0 cp -v --no-preserve=mode --target-directory "$out" --parents

      for dtb in $(find "$out" -type f -name '*.dtb'); do

        ${flip (concatMapStringsSep "\n") overlays (o: ''
          echo -n "Applying overlay ${o.name} to $(basename "$dtb")... "
          mv "$dtb"{,.in}

          # dtmerge requires a .dtbo ext for dtbo files, otherwise it adds it to the given file implicitly
          dtboWithExt="$TMPDIR/$(basename "${o.dtboFile}").dtbo"
          cp -r ${o.dtboFile} "$dtboWithExt"

          echo XXX ${concatStringsSep " " (mapAttrsToList (name: value: "${name}=${value}") (paramsPerOverlayMap.${o.name} or { }))}

          dtmerge "$dtb.in" "$dtb" "$dtboWithExt" ${concatStringsSep " " (mapAttrsToList (name: value: "${name}=${value}") (paramsPerOverlayMap.${o.name} or { }))}

          echo "ok"
          rm "$dtb.in" "$dtboWithExt"
        '')}

      done'';
}))
```

```nix
{
  flake.nixosModules.dtmerge =
    { config, lib, pkgs, ... }:

      with lib;

      let
        overlayParamsType = types.submodule {
          options = {
            name = mkOption {
              type = types.str;
              description = lib.mdDoc ''
                Name of an overlay for which params are going to be passed to dtmerge.
              '';
            };

            params = mkOption {
              type = types.attrsOf types.str;
              description = lib.mdDoc ''
                Params to pass to dtmerge.
              '';
            };
          };
        };

        paramsPerOverlayMap = builtins.mapAttrs
          (_: value: builtins.foldl' (a: b: a // b) { } (builtins.catAttrs "params" value))
          (builtins.groupBy (x: x.name) config.hardware.deviceTree.overlaysParams);

        dtmergeOverlay = final: prev: {
          deviceTree.applyOverlays = break (final.callPackage ((import ./applyOverlays.nix) paramsPerOverlayMap) { });
        };
      in
      {
        options = {
          hardware.deviceTree.overlaysParams = mkOption {
            default = [ ];
            type = types.listOf overlayParamsType;
          };
        };

        config = {
          nixpkgs.overlays = [
            dtmergeOverlay
          ];
        };
      };
}
```

Ramblurr, 2023-09-04:
> see https://bpa.st/raw/CJUPI for the nixpkgs changes
> https://bpa.st/raw/QHONW for the nixos-hardware changes

@justinkb those pastebin links are dead now :( could you re share your patches please? I'm also trying to get access to more dtparams for my NixOS powered rpis.

justinkb, 2023-09-04:
I'm just about to go to bed, I'll try to see if I still have that stuff
saved somewhere tomorrow. If not, shouldn't be too hard to recreate it.
It's a relatively small edit of the stuff in this repo

closed 2026-08-16 by doronbehar