NixOS/nixos-hardware #279: Issues with connecting nixos-21.05 on a raspberry-pi-4 to multiple LG monitors:

zeta-00, 2021-06-23:
Hello there,
The pi connects fine to a single LG monitor, xrandr detects it, and I'm able to do customizations with it(except, I'm unable to change brightness with xrandr, so I have to manually press the monitors buttons to set the brightness). But, whenever I try to connected the pi to a 2nd LG monitor(which is the exact same model), and it stays stuck in the rainbow screen? And xrandr does not detect the 2nd monitor either? I enabled home-manager's autorandr option, but that does not fix this issue either?
I'm posting this issue here, because I'm not sure if it's a hardware problem.
Also I haven't messed with the pi's bootloader config.txt file, so I don't know if that would fix this issue.
If you could at least give some tips on how to troubleshoot this issue.
Thanks in advance for helping fix this issue.

zeta-00, 2021-06-24:
Ok, I've made some progress, I'm guessing that I'm at the point where I need to tweak the pi's bootloader, but I'm hesitatant to make changes there, because I don't want to brick the pi, making it unusable.

https://www.reddit.com/r/NixOS/comments/o5cj36/issues_getting_multiple_monitors_working_with_a/h2vx9jt/?context=3

zeta-00, 2021-06-26:
Here's the same issue with more information:

https://www.reddit.com/r/NixOS/comments/o82k11/nixospi4_issues_trying_to_get_a_dualdisplay_setup/

Mic92, 2021-06-26:
> Ok, I've made some progress, I'm guessing that I'm at the point where I need to tweak the pi's bootloader, but I'm hesitatant to make changes there, because I don't want to brick the pi, making it [...]

It is unlikely that this change would be permanent if you just edit some text files, you can mount the sdcard on a different machine and change it back.

zeta-00, 2021-06-28:
@Mic92 , the changes do seem to be permanent when I mount and configure the config.txt file directly on the pi.

zeta-00, 2021-06-28:
@Mic92 , I'm able to get it to render to two 2k monitors, but ran into more issues:
https://github.com/NixOS/nixos-hardware/issues/282

zeta-00, 2021-06-28:
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

#fix for rpi 4 models with dual hdmi
hdmi_force_hotplug:0=1
hdmi_force_hotplug:1=1
hdmi_drive=2

zeta-00, 2021-06-28:
I've had the best luck with this:
#fix for rpi 4 models with dual hdmi
hdmi_force_hotplug:0=1
hdmi_force_hotplug:1=1
hdmi_drive=2