NixOS/nixos-hardware #288: Bad source for surface go wifi

ursi, 2021-07-05:
https://github.com/NixOS/nixos-hardware/blob/41775780a0b6b32b3d32dcc32bb9bc6df809062d/microsoft/surface/firmware/surface-go/ath10k/ath10k-replace.nix#L5

^ this doesn't exist

https://github.com/NixOS/nixos-hardware/blob/41775780a0b6b32b3d32dcc32bb9bc6df809062d/microsoft/surface/repos.nix#L10-L15

https://github.com/kvalo/ath10k-firmware/tree/84b47062aab31d67156e0a3ef593a6999a12864b

samueldr, 2021-07-05:
cc @mexisme who contributed this.

mexisme, 2021-07-05:
Thanks.
Will try to look tonight; I have an idea what may have happened.

ursi, 2022-02-06:
I take it this has been fixed?

Mic92, 2022-02-20:
> I take it this has been fixed?

see https://github.com/NixOS/nixos-hardware/pull/374 maybe you can push the fix over the finish line.

Mic92, 2022-02-25:
fixed.

closed 2022-02-25 by Mic92