NixOS/nixos-hardware #289: Surface Pro 4 touchscreen not working, kernel update should fix

CTech-BentonE, 2021-07-18:
I'm getting this issue exactly: https://github.com/linux-surface/linux-surface/issues/407. Based on the last two comments in that thread, it looks like the solution is to update the kernel. We are on 5.10.19 and they reported resolution when going from 5.10.x to 5.11.x. 

@mexisme can you do this?

BentonEdmondson, 2021-07-18:
Meant to post that with this account

mexisme, 2021-07-20:
> I'm getting this issue exactly: [linux-surface/linux-surface#407](https://github.com/linux-surface/linux-surface/issues/407). Based on the last two comments in that thread, it looks like the solution is to update the kernel. We are on 5.10.19 and they reported resolution when going from 5.10.x to 5.11.x.
> 
> @mexisme can you do this?

I can look into upgrading the kernel version tonight.
It looks like the latest patches target 5.12.14, so I'll try for that.

devins2518, 2021-07-23:
@mexisme any status on this? I've been facing some issues where shutdown hangs after a suspend so I'm hoping this update might fix it

mexisme, 2021-07-24:
> @mexisme any status on this? I've been facing some issues where shutdown hangs after a suspend so I'm hoping this update might fix it

I'm working on it.  (The job has taken a lot of my time over the last month).

I have a custom 5.11 working on my Surface Go, but that has the `iptsd` settings removed to make it work, so might not be appropriate.
I don't have access to any other Surface hardware any longer, so it might need some testing from others before it can be merged.

closed 2021-08-08 by Mic92

Flameopathic, 2024-05-09:
I am having this issue on my Surface Pro 7. I am working on figuring out a fix for it, and my current idea is to try the version of iptsd on `unstable`, but there is no way that i can find to override `nixpkgs`, which might continue to be an issue because `services.iptsd` does not have a `package` option. thoughts?

PR #300 (2021-08-08): linux_surface kernel 5.10.19->5.13.4
cc: @mexisme

Updated kernel version to 5.13.4. I've been able to build and run it for a few hours and it's working fine on my Surface Pro 4 so far. Haven't tested a lot but it should work fine. 

Should fix #289. cc: @BentonEdmondson