NixOS/nixos-hardware #296: Proposal: tracking of printers and different hardware

zarelit, 2021-07-25:
I found that sometimes the information you need to successfully use a device such as a printer or other kinds of external devices is buried in the wiki or in comments in nixpkgs.

I was thinking that we could have profiles that are named like the commercial names of the devices (e.g. HP Deskjet xyz) and map to their specific drivers and configurations.

What do you think about it?

There are still some open questions like:
* what to do when there is a proprietary driver and an open one that are different feature-wise?
* how do we manage things like "enable airscan" to deal with network scanners?

Mic92, 2021-07-25:
Some of the hardware could be also added to modules/hardware in nixpkgs itself. 
Especially if it comes with packages required, since nixos hardware does not have a good way of auto-updating packages.
In nixos-hardware we usually have subprofiles to switch between different flavors i.e. different drivers.

zarelit, 2021-07-25:
@Mic92 If there is a need for special support such as systemd units the way is to add a `hardware` module in nixpkgs, but I was  specifically thinking about lots of printers that already have the drivers packaged, more often than not it's a matter of mapping the device name with their driver, e.g. canon pixma tr4550 means that you have to add `services.printing.drivers = [ pkgs.cnijfilter2 ];` for printing and `hardware.sane.extraBackends = [ pkgs.sane-airscan ];` for scanning wireless.

Good that we have subprofiles, I'll look into them and hopefully have the time to open a first PR :-D

Mic92, 2021-07-25:
Ok. Mapping drivers from nixpkgs seems like a fair idea.

Mic92, 2021-07-25:
Here an example for subprofiles: https://github.com/NixOS/nixos-hardware/tree/master/dell/xps/15-9560

turion, 2022-11-28:
This sounds like a great idea!

l0b0, 2024-01-16:
Absolutely! I've spent about three hours now trying to configure an HP printer (never buying from them again), and after about 20 iterations of the URL (using about six different URL schemas!) it finally actually prints. It ignores the `InputSlot` and `HPPJLColorAsGray` settings, but it prints. So far:

```
{
  hardware.printers = {
    ensurePrinters = [
      {
        name = "HP_Color_LaserJet_MFP_M281fdn_4FDF62";
        location = "Upstairs workroom";
        deviceUri = "hp:/net/HP_ColorLaserJet_MFP_M278-M281?hostname=${(pkgs.lib.lists.findSingle (host: builtins.elem "printer" host.value) (throw "none") (throw "multiple") (pkgs.lib.attrsets.attrsToList config.networking.hosts)).name}";
        model = "HP/hp-color_laserjet_mfp_m278-m281-ps.ppd.gz";
        ppdOptions = {
          Collate = "False";
          Duplex = "DuplexNoTumble";
          HPOption_Duplexer = "True";
          HPPJLColorAsGray = "True";
          InputSlot = "Auto";
          MediaType = "Plain";
          PageSize = "A4";
        };
      }
    ];
    ensureDefaultPrinter = (builtins.elemAt config.hardware.printers.ensurePrinters 0).name;
  };
}
```