NixOS/nixos-hardware #299: Lenovo T14s: Looking for solution to control microphone mute

devhell, 2021-07-30:
Everything seems to be working fine, except for microphone mute. I've tried a number of things, including `xbindkeys` to get it to mute (and light the LED when muted), and I've had some success, but it stopped working now for whatever reason. So, I thought that I'd ask here if anyone has a more reliable solution before I try to go down a rabbit hole again.

Mic92, 2021-07-30:
You can use `pamixer`: `pamixer --default-source --mute`. I use it like this in systemd services: https://github.com/Mic92/dotfiles/blob/b97ab49c2b50ccdae2b136d66a920941d932bbec/nixos/turingmachine/configuration.nix#L126

closed 2021-07-30 by Mic92

wmertens, 2021-07-30:
For what it's worth, I have a yoga and the mute button just works, no config. I'm using gnome though.

devhell, 2021-07-30:
Thanks @Mic92, I should have mentioned that it's the AMD version, and I tried that already and it doesn't work.

Thank you @wmertens, I'm sure it would be fine if I'd use a full-blown DE, but I'm using i3.