NixOS/nixos-hardware #311: Does the dell e7240 profile or readme need some updating?

danwdart, 2021-09-08:
Having some suspend issues myself but I'm not about to go all the way back to 4.14 kernels when the 5.xx upstreams have fixed it somewhere.

Mic92, 2021-09-09:
Potentially out-dated, any thoughts? @ashgillman
I cannot comment on that further as I don't own the hardware.
In general you can try `linuxPackages_latest` to see if it fixes the issue.

ashgillman, 2021-09-09:
I'm afraid I'm no longer using the hardware to comment with any certainty.

However, the linked bug report in the README reports the problem as fixed in 4.18.10, I can't comment on 5.xx but if it works for you without adding this line then I suggest the README be removed.

I imagine [this file](https://github.com/NixOS/nixos-hardware/blob/master/dell/e7240/default.nix) to still be fine.

danwdart, 2021-09-14:
Okay, thanks, maybe I'll manage this one.