NixOS/nixos-hardware #314: lenovo_fix.py is failing

colemickens, 2021-09-18:


colemickens, 2021-09-18:
```
-- Journal begins at Wed 2021-08-18 02:11:48 PDT, ends at Sat 2021-09-18 02:04:28 PDT. --
Sep 18 02:02:49 xeep systemd[1]: lenovo_fix.service: Failed with result 'exit-code'.
Sep 18 02:02:49 xeep systemd[1]: lenovo_fix.service: Main process exited, code=exited, status=1/FAILURE
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]: FileNotFoundError: [Errno 2] No such file or directory: 'modprobe'
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:     raise child_exception_type(errno_num, err_msg, err_filename)
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:   File "/nix/store/dqxic3j7csd4ywn94n4smmnz55p039g3-python3-3.9.6/lib/python3.9/subprocess.py", line 1821, in _execute_child
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:     self._execute_child(args, executable, preexec_fn, close_fds,
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:   File "/nix/store/dqxic3j7csd4ywn94n4smmnz55p039g3-python3-3.9.6/lib/python3.9/subprocess.py", line 951, in __init__
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:     with Popen(*popenargs, **kwargs) as p:
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:   File "/nix/store/dqxic3j7csd4ywn94n4smmnz55p039g3-python3-3.9.6/lib/python3.9/subprocess.py", line 349, in call
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:     retcode = call(*popenargs, **kwargs)
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:   File "/nix/store/dqxic3j7csd4ywn94n4smmnz55p039g3-python3-3.9.6/lib/python3.9/subprocess.py", line 368, in check_call
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:     subprocess.check_call(('modprobe', 'msr'))
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:   File "/nix/store/qs15wsc7l4b35ywaciv49cm8vqg7vhym-throttled-0.9.2/bin/.lenovo_fix.py-wrapped", line 278, in set_msr_allow_writes
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:     set_msr_allow_writes()
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:   File "/nix/store/qs15wsc7l4b35ywaciv49cm8vqg7vhym-throttled-0.9.2/bin/.lenovo_fix.py-wrapped", line 916, in main
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:     main()
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]:   File "/nix/store/qs15wsc7l4b35ywaciv49cm8vqg7vhym-throttled-0.9.2/bin/.lenovo_fix.py-wrapped", line 992, in <module>
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]: Traceback (most recent call last):
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]: [I] Trying to unlock MSR allow_writes.
Sep 18 02:02:49 xeep lenovo_fix.py[1133227]: [I] Detected CPU architecture: Intel Kabylake
Sep 18 02:02:49 xeep systemd[1]: Started Stop Intel throttling.
```

jokogr, 2021-09-18:
On my machine (ThinkPad T480s on NixOS 21.05 with Python 3.8 and throttled 0.8) it seems to be working.

My guess is that the kernel module `msr` is not loaded and then lenovo_fix.py tries to load it with `modprobe` ([reference](https://github.com/erpalma/throttled/blob/d24047a2824713c647a58a177d7a4db1e28e0a1f/lenovo_fix.py#L277)), which is not in the path (it would also need root privileges).

Before patching the service, could you try to load the `msr` kernel module on your own, e.g., via `boot.kernelModules`?

colemickens, 2021-09-18:
That does solve the issue, I feel like we should:

1. fix this script
2. maybe more importantly, add `msr` to the dell xps module(s)?

jokogr, 2021-09-18:
I vouch for the second suggestion, it seems to me the least intrusive one.

@colemickens since you mentioned Dell XPS, I assume that you are using the 13-9370 module. @michaelpj as you were the one who enabled throttled in 2bf94227c1efbf47e99f55dacd799fffe87207e8, did you have any issues with the `msr` kernel module?

colemickens, 2021-09-18:
I'm down to do both, I don't know why we'd bother leaving the modprobe in there if it will never be useful? Or is it more about being able to use the upstream python script as is?

michaelpj, 2021-09-19:
Hmmm. I'm not sure what the deal is here. I actually turned off `throttled` myself since I didn't seem to be getting much benefit from it. I definitely didn't have to do anything with `msr`, perhaps I just had it enabled by default? Possibly the nixos module should enable the kernel module if it needs it.

jokogr, 2021-09-20:
@michaelpj thanks for replying! You're right, I also think that the `msr` module is loaded automatically by the kernel, at least this is the behavior I see on all my systems, too.

Nevertheless, I guess it does not hurt enabling it explicitly, so @colemickens feel free to make a PR on that.

colemickens, 2021-09-20:
Strange, I was doing upgrades, but I certainly never went out of my way to unload `msr` or anything. Really no idea how I'd wind up there if `msr` is supposedly auto-loaded?