NixOS/nixos-hardware #316: System powering off out of the blue

wmertens, 2021-09-24:
In recent kernels (a month or two), whenever my system gets really busy, it decides to suspend to disk.

Unfortunately I don't have suspend-to-disk set up (I don't have a swap partition, just several swap files), so it just powers off. 

Any idea what caused this, and what I might do to stop this from happening?

I have this config:

```
  powerManagement.cpuFreqGovernor = lib.mkDefault "powersave";
  services.thermald.enable = true;

  # Use the power daemon thingie in Gnome
  services.tlp.enable = false;
  services.tlp.settings = {
    CPU_SCALING_GOVERNOR_ON_AC = "powersave";
    CPU_SCALING_GOVERNOR_ON_BAT = "powersave";
  };
```

Mic92, 2021-09-25:
Does the CPU becomes hot before this?

wmertens, 2021-09-27:
@Mic92 :thinking: indeed I have a Lenovo C940 and I found this issue https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1873083 but thermald is version 2.4.3 now.

Also, it didn't happen at all until a month or two ago, so it must be something software related. Thanks for the pointer!

wmertens, 2021-09-27:
I tried unloading the intel module like described in the issue but that didn't help.

The system log only has this to say:

```
Sep 27 15:11:52 wmertens-nixos wpa_supplicant[1361]: wlp0s20f3: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-64 noise=9999 txrate=433300
Sep 27 15:11:52 wmertens-nixos systemd-logind[1085]: Power key pressed.
Sep 27 15:11:53 wmertens-nixos .gnome-shell-wr[2184]: Run-or-raise> No listeners [action={}], 164
Sep 27 15:11:53 wmertens-nixos ModemManager[1689]: <info>  [sleep-monitor] system is about to suspend
```

So it looks like the BIOS thinks the CPU is getting too hot and turns off the system by pretending to press the power button. :-/