NixOS/nixos-hardware #317: Adding support for building installers?

samueldr, 2021-09-27:
More of a question for other maintainers, to see if there are strong reasons against.

I'm currently adding an ARM device to `nixos-hardware`, and have been thinking it would be nice to be able to build the NixOS installer images (whether SD image, or UEFI iso) with the profile baked-in.

With ARM devices, it would be sometimes necessary, as sometimes the kernel requires tweaks and such nonsense.

With your run-of-the-mill x86_64 system it might also be helpful. I'm thinking about those laptops with *annoying to setup Wi-Fi*, or maybe other "fun" quirks requiring "fun" configuration.

* * *

There's another dimension to this question: would it be acceptable to also provide some sort of "harness" to make cross-compilation easier? I'm thinking about the "first time setup" for a local AArch64 system, when the user has no AArch64 hardware to use to build that installer system. Basically slipstream [part of what I already have here](https://github.com/samueldr/cross-system/) so it's easier for end-users.

samueldr, 2021-09-27:
@Mic92 any opinion for or against?

Mic92, 2021-10-09:
I think providing installer would still fit into the scope of this project.
I am just not sure how we would provide the CI to make this happen.
Images itself can be hosted in  a github release as I demonstrate here: https://github.com/nix-community/nixos-images/releases/tag/nixos-unstable