NixOS/nixos-hardware #323: Framework Support

Emiller88, 2021-10-18:
I think getting support would be cool opening an issue for tracking.

https://community.frame.work/t/using-the-ax210-with-linux-on-the-framework-laptop/1844/65

Setting the kernel to 5.12.15 for now for example rather than recommending latest.

samueldr, 2021-10-18:
Not exactly trivial considering it's EOL and out of Nixpkgs already.

There may be another workaround for the issues with the ax210, which requires deleting a firmware from the firmware package, which is *more trivial than expected* with an overlay. But I wasn't able to get any of the two individuals I know with the Framework laptop to confirm the workaround I expect to work would work.

If anyone has the Framework laptop and has time to check this out, it can be checked.

Emiller88, 2021-10-19:
I just threw this in there and it appears to be working for me

```nix
  boot.kernelPackages = pkgs.linuxPackagesFor (pkgs.linux_5_10.override {
    argsOverride = rec {
      src = pkgs.fetchurl {
        url = "mirror://kernel/linux/kernel/v5.x/linux-${version}.tar.xz";
        sha256 = "sha256-R/zqmWwMAeWKxfhS/Cltd6NJbFUPMDQVL+536Drjj9o=";
      };
      version = "5.12.15";
      modDirVersion = "5.12.15";
    };
  });
```

Mic92, 2021-10-21:
cc @grahamc and @kvark
who wrote blog articles about the hardware.

conorreid, 2021-11-29:
Using the [latest unstable NixOS](https://hydra.nixos.org/build/158009668) with the following in the `hardware-config.nix` works just fine for me on Framework. Granted I don't think bluetooth is working but that's not a showstopper for me. I could upload some instructions for how I got the AX210 working if there's interest.

```boot.kernelPackages = pkgs.linuxPackages_latest;```

rehno-lindeque, 2021-12-17:
Worth linking those two blog articles here I think

* https://grahamc.com/blog/nixos-on-framework
* http://kvark.github.io/linux/framework/2021/10/17/framework-nixos.html

conorreid, 2021-12-18:
> Using the [latest unstable NixOS](https://hydra.nixos.org/build/158009668) with the following in the `hardware-config.nix` works just fine for me on Framework. Granted I don't think bluetooth is [...]

Following up here, the only additional thing I had to do, besides making sure the line above is included in the `hardware-configuration.nix` file, was set up a `wpa_supplicant.conf` file, stop wpa_supplicant, and restart it with the configuration file. Once I did this, I was connected to wifi in the NixOS installer, and then could proceed with installation of NixOS as normal using the standard manual. Currently writing this on NixOS 21.11 with KDE on my Framework, and it's all working great!

neumantm, 2021-12-21:
Has anyone else got the problem that no disk is mounted when resuming from ram on the framework?

pascaldornfeld, 2022-02-25:
> Has anyone else got the problem that no disk is mounted when resuming from ram on the framework?

I got the same issue right now (probably). After suspend sddm doesnt work and I cannot login even in the terminal. journalctl has no new logs after resume; probably disk not mounted as you said

pascaldornfeld, 2022-02-26:
> > > Has anyone else got the problem that no disk is mounted when resuming from ram on the framework?
> > 
> > 
> > I got the same issue right now (probably). After suspend sddm doesnt work and I cannot login even in the terminal. journalctl has no new logs after resume; probably disk not mounted as you said
> 
> Is this maybe related to https://wiki.archlinux.org/title/Framework_Laptop#Suspend

This is possible. I am using a crucial p5 plus 1tb. The ssd can be the cause for the issue?! Any known fix?

I checked the support page. There are no firmware updates for this ssd.
I am running nixos, not windows.

Mic92, 2022-02-26:
In one case it was possible to disable power management in bios for the ssd: https://community.frame.work/t/laptop-overheating-and-losing-connection-to-m-2-drive/10508/30

neumantm, 2022-02-28:
> In one case it was possible to disable power management in bios for the ssd: https://community.frame.work/t/laptop-overheating-and-losing-connection-to-m-2-drive/10508/30

I think the mentioned Power Plan setting is not a bios setting, but a windows 11 setting.
Does anyone know the equivalent linux configuration option?

closed 2022-03-09 by Mic92

magnetophon, 2023-02-22:
Would it make sense to integrate https://github.com/ssddq/fw-ectool?
That enables control of lights, fan, battery levels etc.

Emiller88, 2023-02-22:
@magnetophon What are the pros and cons?

magnetophon, 2023-02-22:
Without it, you cannot get or set the fan speed, that's the main thing.
Also, it enables finer control of the keyboard backlight, and setting the color and brightness of the power led and the leds on the sides.
 
There's a whole list of low level things you can do: https://www.howett.net/posts/2021-12-framework-ec/ 
We should also integrate https://github.com/TamtamHero/fw-fanctrl for enabling different fan profiles.

I don't now of any downsides, but that's not saying there aren't any.
I only just started using this.
Pinging users that have this in their config: @tarantoj @mdvmeijer 
and upstream: @ssddq

magnetophon, 2023-02-23:
The author of the fw-ectool nix pkg has given me possible downsides: https://github.com/ssddq/fw-ectool/issues/1
Maybe we can discuss it further there?

mdvmeijer, 2023-02-23:
@magnetophon Thanks for the ping!

My reason for integrating @ssddq's nix package was to get https://github.com/TamtamHero/fw-fanctrl working as a systemd service on NixOS. I did it in a quick and dirty way though, since I wanted to use NixOS as my daily driver asap and the fan noise of my Framework bugged me quite a bit.

I've been getting familiar with NixOS/Nix for just over a month, and I wanted to try my hand at creating a nix package for https://github.com/TamtamHero/fw-fanctrl as an exercise soon. Maybe other people will find it useful as well.

magnetophon, 2023-02-23:
@mdvmeijer You're welcome!  :)

I just added a shortcut for finer keyboard backlight control; I need it the most when it's dark, but the default minimum brightness blinds me then.
https://github.com/magnetophon/.dot/commit/ba9443d50199d495fa55c70e16015dcd9336ea91
https://github.com/magnetophon/nixosConfig/commit/c6a3939e71b7f04157d5367d854b1bb04ed036ad

Next I intend to add one that toggles the fans to run quicker, for when I need maximum cpu power, not even kidding!

mdvmeijer, 2023-02-26:
@magnetophon Oh neat! I've run into the same problem with the keyboard backlight brightness, so I'll definitely take a look at your work. And the keyboard shortcut for toggling fan speed reminds me of the turbo button on some oldschool PC's :D 
Before my time, but always thought it was a cool concept.

For anyone interested, I've finished the first version of the fw-fanctrl Nix package and service:
https://github.com/mdvmeijer/fw-fanctrl-nix

For now I'm using https://github.com/ssddq/fw-ectool as a submodule.

magnetophon, 2023-02-26:
In case anyone's interested, I finished the fan toggle: https://github.com/magnetophon/.dot/commit/2d8d7dd5821cb0283a3a1bc35932355347d3f768

thomaseizinger, 2023-12-20:
> Would it make sense to integrate [ssddq/fw-ectool](https://github.com/ssddq/fw-ectool)?

I'd like to bring this discussion back up. I do think that the concerns raised in https://github.com/ssddq/fw-ectool/issues/1#issuecomment-1441119344 have merit. Personally, I don't think that they should be a blocker for _packaging_ the software. However, I am also new to the nix ecosystem so I am not sure if there is precedent for a package that can cause hardware damage when misused. It appears to me that it is a case of RTFM.

I would be very useful to have better out-of-the-box support for controlling laptop's fan speed!

Emiller88, 2023-12-20:
> Personally, I don't think that they should be a blocker for packaging the software. However, I am also new to the nix ecosystem so I am not sure if there is precedent for a package that can cause hardware damage when misused. It appears to me that it is a case of RTFM.

Yeah, it's not a blocker. But it's probably easier on the select people that have found that repo or this issue, have done their research and understood what they're diving it. That's the beauty of Flakes! 😄

PR #372 (2022-03-09): feat(framework): Add initial support
- Enable deep sleep
- Use kernel that's 5.12 or latest
- Enable fingerprint scanner
- HiDPI

closes #323