NixOS/nixos-hardware #338: Provide NVIDIA non PRIME profile

bryanasdev000, 2021-11-20:
I have been using nixos-hardware for quite some time with the common profiles, I have recently changed from an AMD to NVIDIA GPU and the first thing that I did was trying to use the NVIDIA profile, but it expects a PRIME environment, which in my desktop, I do not have.

Would be cool to have a `nixos-hardware/common/gpu/nvidia-nonprime.nix` with:

```
{ lib,  ... }:

{
  # For desktops
  # Non-Optimus mode
  # You need an MXM card
  services.xserver.videoDrivers = lib.mkDefault  [ "nvidia" ];
 
  # For high res tty
  hardware.nvidia.modesetting.enable = lib.mkDefault true;
 
  # Suspend/hibernate
  hardware.nvidia.powerManagement.enable = true;
}
``` 

Ref:

- https://nixos.wiki/wiki/Nvidia
- https://nixos.org/manual/nixos/stable/#sec-x11-graphics-cards-nvidia

Mic92, 2021-11-20:
@bahaynes @thiagokokada opinions?

thiagokokada, 2021-11-20:
I don't see why not. I just don't like the name `nvidia-nonprime`, I think maybe it should have been `nvidia` and the old one should be called `nvidia-prime`, however this would break other people configs :disappointed:.

bahaynes, 2021-11-20:
I think it would make sense. I suspect the reason it hasn’t come up previously is that there aren’t as many common models of desktops using nixos-hardware.

It would be good to move the nvidia files into a folder to match the way we have amd, but that will mean we need to fix the paths in any of the devices that use it currently.

bryanasdev000, 2021-11-20:
> I don't see why not. I just don't like the name `nvidia-nonprime`, I think maybe it should have been `nvidia` and the old one should be called `nvidia-prime`, however this would break other people configs disappointed.

Well I vote for that, just did not want to refactor something that I can not test lol

bryanasdev000, 2021-11-20:
> I think it would make sense. I suspect the reason it hasn’t come up previously is that there aren’t as many common models of desktops using nixos-hardware.
>
> It would be good to move the nvidia files [...]

Yeap, also we may need to have more than a `nvidia.nix` (or make an override?) for legacy drivers (legacy_470, legacy_390, legacy_340) and maybe betas?

thiagokokada, 2021-11-21:
> I think it would make sense. I suspect the reason it hasn’t come up previously is that there aren’t as many common models of desktops using nixos-hardware.
>
> It would be good to move the nvidia files [...]

One proposal:

- Create a new `common/gpu/nvidia` directory, move `common/gpu/nvidia.nix` -> `common/gpu/nvidia/prime.nix` and `common/gpu/nvidia-disable.nix` -> `common/gpu/nvidia/disable.nix`
- Keep the old files, mark them as deprecated, and import the new files respectively
- Add a new `common/gpu/nvidia/default.nix`, that is the non-PRIME profile

WDYT @bryanasdev000 @bahaynes ?

bryanasdev000, 2021-11-21:
> One proposal:
>
> * Create a new `common/gpu/nvidia` directory, move `common/gpu/nvidia.nix` -> `common/gpu/nvidia/prime.nix` and `common/gpu/nvidia-disable.nix` -> `common/gpu/nvidia/disable.nix`
> * [...]

Sounds like a plan for me, it also leave space if in the future we need a `common/gpu/nvidia/legacy470.nix` with overrides for legacy drivers.

About the possible future refactor, a good `sed` does handle it?

bryanasdev000, 2021-12-22:
People, if everyone agrees, I can start working on a PR.

bahaynes, 2021-12-22:
Sounds good to me. I can help with fixing the devices I’ve got.

TLATER, 2022-11-13:
No idea if github notifications ping on PRs, but I implemented @thiagokokada 's suggestion. Would be happy to see a review, I'd like to use these modules :)

commit 7e60458d (2022-11-22): common/gpu/nvidia*: Migrate to common/gpu/nvidia/* and add non-prime

Fixes #338
PR #502 (2022-11-22): common/gpu/nvidia*: Migrate to common/gpu/nvidia/* and add non-prime
###### Description of changes

Fixes #338

Annoyingly, the flake module can't be deprecated cleanly. Suggestions welcome.

###### Things done

Note I can't properly test changes that involve prime/disable, so I'd appreciate confirmation that that works, as simple as this change looks. Also can't test the module changes, of course.

- [ ] Tested the changes in your own NixOS Configuration
- [x] Tested the changes end-to-end by using your fork of `nixos-hardware` and
      importing it via `<nixos-hardware>` or Flake input
closed 2022-11-22 by bors[bot]